prusa3d/Prusa-Firmware-Buddy #1037: [BUG] Misspelling found

3DFans, 2020-11-06:
Just a little thing that can be fixed quickly. 
Because the error devil has crept into the calibration menu. 
There it says: "bis der Filament am", but it must read "bis **das** Filament am". 

regards

JustAnother1, 2020-11-07:
"Druck**blech**" ? That should be "Druck**bett**" !Also "Knopf **D**rehen" should be "Knopf **d**rehen"

3DFans, 2020-11-07:
> "Druck**blech**" ? That should be "Druck**bett**" !Also "Knopf **D**rehen" should be "Knopf **d**rehen"

The pressure plate is already correct because it means the edition. ;) Print bed would be the whole.

JustAnother1, 2020-11-07:
@3DFans If with "edition" you mean the removable print surface / the PEI coated sheet. Then calling that a "Blech" is correct. But as this test is executed with the print surface being placed onto of the print bed it is correct to refer to the print surfaces in the assembled situation as the print bed.
As a native speaker I still think that "Druck**bett**" is the better word here.
But consistency is more important than the best word here. The printer should always refer to the print bed with the same words. If that is the "Blech" than so be it. It is better if the user smiles about the "Blech" than the user being confused what the printer wants now.

JohnnyDeer, 2020-11-09:
Hi @3DFans thank you for reporting.

Chips83, 2020-11-13:
There are to many punctuation marks (,) and some words are in strange places for a native german speaker. I think it would be better to write "Sobald der Drucker beginnt Kunststoff zu extrudieren, stellen Sie die Höhe der Düse durch drehen des Knopfes ein, solange bis das Filament am Druckblech haftet."

closed 2024-07-04 by github-actions[bot]