prusa3d/Prusa-Firmware-Buddy #1038: [FEATURE REQUEST]<Enter comprehensive title>

oopsitsaflame, 2020-11-07:
Hi,

a while ago i tried to add a hardened steel nozzle to my mini. And failed miserably. As most of you know, steel conducts heat way slower than brass, so it needs to be higher in temp to make this up.

Any time i tried loading filament the extruder grinded into it because the pre defined temps for filaments were to low and it could not be molten fast enough.

JohnnyDeer, 2020-11-19:
Hi @oopsitsaflame thank you for suggestion. We noted this as possible future feature.

RichiH, 2020-12-11:
As Prusa released blog posts, live video, etc around food-safe plastics and the emphasis on using non-brass nozzles due to their lead content, is this a confirmed feature now?

SM-26, 2021-01-11:
Dumb question, but I've now added a Harden steel nozzle.
If we do a PID tune, and add the value in the start of gcode, should this be negated?

oopsitsaflame, 2021-01-11:
> Dumb question, but I've now added a Harden steel nozzle.
> If we do a PID tune, and add the value in the start of gcode, should this be negated?

no it should not, the thermistor is in the heating block

closed 2024-07-04 by github-actions[bot]