prusa3d/Prusa-Firmware-Buddy #1052: [BUG] Printer Status on GUI does not update on Pause

JeffersGlass, 2020-11-13:
**Printer type** - [MINI]

**Printer firmware version**-  4.2.0

**Original or Custom firmware** Original

**Optional upgrades** - None

**USB drive or USB/Octoprint**
USB Drive

**Describe the bug**
When the printer is printing, clicking the "Pause" button on the GUI does not change the "Printing" header at the top of the GUI, nor the associated print icon. The header remains "Printing" with the triangular "printing" symbol next to it.

**How to reproduce**

1. Begin a new print from USB.
2. Once print has started, pause the print.
3. Observe that the header and icon still reflect the "printing" state rather than the paused state

**Expected behavior**
The header should change to "Paused" with the pause icon, or perhaps should change to "Pausing" and then "Paused"

**Video**
[IMG_2149 (1).zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5537783/IMG_2149.1.zip)

JohnnyDeer, 2020-11-18:
Hi @JeffersGlass thank you for reporting and also big thanks for your pull request :)

DRracer, 2020-11-25:
@JeffersGlass your PR has been approved and merged (will be in the upcoming FW 4.3-RC1), would you mind closing this issue?

JeffersGlass, 2020-11-25:
Can do! Thanks for integrating the fix.

closed 2020-11-25 by JeffersGlass