prusa3d/Prusa-Firmware-Buddy #1062: [FEATURE REQUEST] Display time till Filament change

AlexW-578, 2020-11-17:
When making a multiple colour print on the Mini, it would be nice if you could see how long you have until you will need to change the filament.
This could be done on either the web interface or as an option to replace the end time with.
Thanks

jkavalik, 2020-11-17:
I believe that would have to be implemented in the slicer mainly, that one generates the timing info which the printer only shows, that could be configured to show time to next change instead time to finish without any firmware change. Showing both at once would probably need both slicer and firmware support.
Would be interesting to have the progressbar show marks for planned changes to :)

deepthought-64, 2020-11-17:
Cool idea, that would be a nice feature. Not only for the MINI but also for the i3 printers! +1

JohnnyDeer, 2020-11-19:
Hi, as @jkavalik wrote this feature is related to both slicer and FW and we have plan to implement it in future to Mini and i3 also.

fortinmike, 2021-02-16:
> Would be interesting to have the progressbar show marks for planned changes to :)

Great idea! Might also be nice to color the progress bar segments with the colors as specified in the slicer :) That or a small color swatch on the progress bar on color changes, or a subtle colored underline. Brain dump:

[image]

(the bright orange part representing the current progress)

ConorD01, 2024-02-26:
Really miss this feature from the mk3s+. Thought it was super useful to see time to filament change at a glance and know when id need to swap filament. Seems an easy thing to add.

closed 2024-07-04 by github-actions[bot]