prusa3d/Prusa-Firmware-Buddy #1065: [ENHANCEMENT] Please remove safety feature for filament load/unload

willbelr, 2020-11-18:
Hi, in MINI firmware version 4.1.0, a safety feature was introduced to "remember information about successfully loaded filament and its type".

However, it is  unpratical, especially when trying to unload stuck filament. Sometimes the unload fails, so the filament must be purged before unloading again, and this feature force to "reload" the filament which may worsen the problem (ie. further grinding filament in the extruder).

The MINI is full of useful features to simplify the printing process, and it is appreciable. However, this takes it a bit too far and is nuisible in my opinion. Could there be a switch to turn that feature off? Thank you.

DRracer, 2020-11-25:
@willbelr the feature will be kept, but the printer will allow overriding it (asks: are you sure you want to load/unload filament? in the upcoming FW 4.3

Hamcha, 2020-12-09:
Thank you so much, this has been bugging me too for a while

JohnnyDeer, 2020-12-14:
Hi @willbelr function described by @DRracer is implemented in 4.3.0-RC1. I close this issue as solved.

closed 2020-12-14 by JohnnyDeer