prusa3d/Prusa-Firmware-Buddy #1067: [BUG] Vastly incorrect time estimate during warmup

reedy, 2020-11-19:
**Printer type** - [MINI]

**Printer firmware version**-  4.2.1-final+2072

**Original or Custom firmware**- Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
USB Drive

**Describe the bug**
During the nozzle/bed heatup time, the software was suggesting that the print was going to take many weeks (26th December) to print, when it should have been less than an hour. Very soon after, it displayed a much more reasonable ~50 minute run time

I've not noticed it on other prints, but I don't watch the printer like a hawk either :)

**How to reproduce**
Using the attached gcode on the prusa mini...

**Expected behavior**
An estimate that isn't completely incorrect

**G-code**
[new-base_0.2mm_PETG_MINI_49m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5567650/new-base_0.2mm_PETG_MINI_49m.gcode.zip)

**Crash dump file**
N/A

**Video**
N/A

[image]
[image]

JohnnyDeer, 2020-12-16:
Hi @reedy, thank you for reporting. Do you have connected ethernet during print? If yes please try it without, thank you.

reedy, 2020-12-16:
I did... But why should that make any difference? Surely it's designed to have the ethernet connected?

I've not noticed it again since (but I've not attempted to re-print the same files either)

JohnnyDeer, 2021-02-17:
> But why should that make any difference?

Should not, but things like factory reset, ethernet connection always worth to try. 
So you don't met this behavior again since?

reedy, 2021-02-26:
I've just had it again now. I'd done one print (about 3 hours), left it for 10 minutes or so then started off a second print. It was telling me it would finish sometime in April. Unforunately it was back to a more reasonable time estimate by the time I'd grabbed my phone

Ethernet was connected...

reedy, 2021-02-27:
[image]

[image]

reedy, 2021-04-03:
[image]

[image]

reedy, 2021-04-03:
Apparently not just during warmup...

Using the attached gcode

[Case_Upper_V5.STL_0.15mm_PETG_MINI_3h32m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/6252533/Case_Upper_V5.STL_0.15mm_PETG_MINI_3h32m.gcode.zip)

reedy, 2021-04-04:
So when you boot without ethernet (so no NTP time, and with #1285 meaning the web interface is useless), it just display a relative time, rather than "print will end at"....

Only just started doing prints like this, so we'll see if it helps or not.

Though, why can't I choose to have it display "print will end at" or "time remaining", especially when it does have system time set via NTP?

Prusa-Support, 2022-12-16:
The new firmware [4.4.1-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1-RC1) improves the estimated end time thanks to the adoption of the RTC (Real Time Clock), and possibly solve this issue.

Could this issue be closed?

Michele Moramarco
Prusa Research

reedy, 2022-12-16:
Would be nice for it to become a stable release and have a chance to test it :)

Eldenroot, 2022-12-16:
I have to try :) Thx!

hreality, 2022-12-24:
What would be really nice (opinion) is if the config we save to the drive could specify a timeserver.

reedy, 2022-12-24:
> What would be really nice (opinion) is if the config we save to the drive could specify a timeserver. 

What is it using? Does it listen to DHCP?

hreality, 2022-12-24:
It’s using the static IP listed but mode is showing DHCP.  No idea how to
change the time server from U to say Central

-- 

*Disclaimer: This message contains confidential information and is intended
only for the individual named. If you are not the named addressee you
should not disseminate, distribute or copy this e-mail. Please notify the
sender immediately by e-mail if you have received this e-mail by mistake
and delete this e-mail from your system. E-mail transmission cannot be
guaranteed to be secure or error-free as information could be intercepted,
corrupted, lost, destroyed, arrive late or incomplete, or contain viruses.
The sender therefore does not accept liability for any errors or omissions
in the contents of this message, which arise as a result of e-mail
transmission.*

reedy, 2023-01-13:
[image][image]

Definitely still present on older versions 😅

Will upgrade when stable

Prusa-Support, 2023-01-24:
Last week the FW 4.4.1 (final), which improves clock accuracy, was released.
We would like to get your feedback before closing this issue.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-02-14:
This issue will be closed as solved now.

Michele Moramarco
Prusa Research

closed 2023-02-14 by Prusa-Support