prusa3d/Prusa-Firmware-Buddy #1074: 5015 Fan MK3s and mini differently

Lino77, 2020-11-20:
It could be that with the current github version the 5015 fan of the MK3s doesn't work properly anymore, it doesn't pass the fan test and doesn't turn at full speed

JohnnyDeer, 2020-11-30:
Hi @Lino77 your issue is not understable please describe more informations or close this one and open new issue following templates thank you.

Lino77, 2020-12-03:
As it looks to me, I have defective fans that should not reach their speed. The delta should normally reach 8000 RPM and the pressure fan 6000 RPM at 100%

Firmware Realese 4.3.0 github

SELFTEST START
printer serial: ###############

Fan0 Started
Fan1 Started
Fan0 at 20% PWM = 906 RPM
Fan1 at 20% PWM = 1364 RPM
Fan0 at 40% PWM = 2221 RPM
Fan1 at 40% PWM = 2946 RPM
Fan0 at 60% PWM = 3350 RPM
Fan1 at 60% PWM = 4410 RPM
Fan0 at 80% PWM = 4113 RPM
Fan1 at 80% PWM = 5439 RPM
Fan0 at 100% PWM = 4664 RPM
Fan1 at 100% PWM = 6344 RPM
Fan0 Passed
Fan1 Passed
SELFTEST END

closed 2020-12-03 by Lino77