prusa3d/Prusa-Firmware-Buddy #1075: [FEATURE REQUEST] Play sound on print finishing

reedy, 2020-11-21:
**Printer type** - MINI

**Describe the expected functionality**
It would be useful (for me, and should be optional) if the printer played a sound/tune when the printing finishes. I'm usually not in the same room as my printer, so can't necessarily hear when it stops printing. If it played a noise (a few beeps) when it was done, that would be helpful and I'm more likely to be able to hear it.

Helpful for those of us that don't have remote monitoring setup (especially with push notifications etc)

reedy, 2020-11-21:
I don't know what happens on error, but similar happening then would be good too...

Gordontgopher, 2020-11-22:
Duplicate of #334

reedy, 2020-11-22:
> Duplicate of #334

Maybe maybe not. I guess that is getting the gcode to make a sound and therefore needs putting into every spliced file etc. I dunno if that is normal/expected (though the fact it doesn't play a sound is presumably a bug). I'm asking for it on a more general case, where it's played every time any print finishes...

hemdrup, 2020-12-20:
Yes. But, please add a way in setup to turn it on/off.

fortinmike, 2021-02-16:
I too would love this feature. It would help to not waste time to start a new print when doing multiple prints in a row.

I would also like an option to either have it beep a few times, OR continuously until the user confirms the print is finished (or interacts with the UI). I'd personally keep the printer in "beep until interacted with" mode so that whenever I get close enough to hear the beep (which could be several minutes after the print ends) I can tend to my printer and start a new print. Kind of like the timer on my oven, which beeps until someone comes to turn it off.

Dev01-D, 2021-03-18:
I know i have had this running on my Ender 3 and it was by using an entry in my end gcode.

fortinmike, 2021-04-03:
I tried using M300 in end gcode in PrusaSlicer but unfortunately it does nothing. I think barring a specific "beep on end" feature being implemented in the firmware just adding support for M300 on the MINI would do the job for many.

matthewlloyd, 2021-06-28:
Anyone following this issue may be interested to know that I just released custom firmware that adds support for `M300` (play a sound), amongst other goodies:

http://github.com/matthewlloyd/Llama-Mini-Firmware#readme

foorschtbar, 2022-09-03:
Any update here? I wait now since 2 years for this simple future...

funtomaas, 2022-12-26:
Hey: at least with the new firmware it is possible to play Sound with Marlin Code M300, so I'm happy with it.

foorschtbar, 2022-12-31:
> Hey: at least with the new firmware it is possible to play Sound with Marlin Code M300, so I'm happy with it.

You are right! It works now - still cant find any information about that in the release notes, but hey, it works after 2 years now!

Eldenroot, 2023-01-01:
G-code start no problem, G-code end bugged.

@Prusa-Support

Prusa-Support, 2023-01-05:
Thanks for testing this on the most recent firmware, and for reporting the problem.
Our developers will look into this.

Michele Moramarco
Prusa Research

foorschtbar, 2023-01-05:
> Thanks for testing this on the most recent firmware, and for reporting the problem. Our developers will look into this.
> 
> Michele Moramarco Prusa Research

Thank u. It look like the ([#1796](https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/1796), [#1800](https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/1800)) not adding the normal M300 Marlin style beeps with duration and frequency [(M300 [P<ms>] [S<Hz>])](https://marlinfw.org/docs/gcode/M300.html), but only a single beep what is a way better than nothing, but far away from good :(

Eldenroot, 2023-01-05:
Better than nothing but should be added somehow if prusa mini uses merlin and uses same documentation

Prusa-Support, 2023-01-20:
Yes, our firmware is based on Marlin but substantially modified. Our firmware doesn't necessarily guarantee to be compatible with all functionalities of a stock-Marlin firmware, **and vice versa**.

I'd keep the issue open to highlight the presence of a bug so our developers will look into this.

Michele Moramarco
Prusa Research

ErGeeBee, 2023-03-14:
I would love this feature integrated in the Buddy UI. With on/of function.

Eldenroot, 2023-11-03:
**just add code for sound into gcode start and end in your slicerr**

MightyPork, 2023-11-03:
> **just add code for sound into gcode start and end in your slicerr**

Like this?

[image]

Prusa-Support, 2023-12-09:
You can adjust parameters like
```
S: Frequency in Hz
P: Duration in ms
V: Volume
```
https://help.prusa3d.com/article/buddy-firmware-specific-g-code-commands_633112

This said, I'm not sure if the current behavior will change but the code `M300` with its parameters seems to work correctly and would address the specific needs of users who would like the printer to hermit a sound in specific circumstances.
This can normally be done via the _custom gcode boxes_ in Prusa Slicer - or editing the gcode to the user's liking.

Can the issue be closed now?

Michele Moramarco
Prusa Research

foorschtbar, 2023-12-10:
> Can the issue be closed now?

Something is still very strange. **The playback seems to be too fast.**

Try this gcodes 

```
;Scooter
M300 S1567 P206
M300 S1567 P206
M300 S1396 P103
M300 S1318 P103
M300 S1396 P413
M300 S1174 P310
M300 S1046 P103
M300 S1174 P206
M300 S1567 P206
M300 S1567 P206
M300 S1396 P206
M300 S1318 P206
M300 S1567 P206
M300 S1567 P206
M300 S1396 P103
M300 S1318 P103
M300 S1396 P413
M300 S1174 P413
M300 S1318 P206
M300 S1046 P206
M300 S1174 P827
M300 S0 P206
```

1. In a webplayer https://foorschtbar.github.io/GCode-Ringtone/ -> sounds right
2. On printer (i uploaded a *.gcode file only with that content to octoprint and "print" the file) -> sounds strange

Prusa-Support, 2024-01-04:
If that is the case, please consider opening a specific separate issue about that.

Michele Moramarco
Prusa Research

wohali, 2024-01-19:
M300 seems not working on MK4 that worked on MK3. This case cannot be closed, @Prusa-Support .

Prusa-Support, 2024-02-20:
It seems to work to me, also on MK4.

What seems to work best though is "duplicating" the duration with a `G4` "[Dwell](https://help.prusa3d.com/article/prusa-firmware-specific-g-code-commands_112173)" code - or even extending it by several milliseconds to generate pauses.

For example 

```
;Scooter
M300 S1567 P206
M300 S1567 P206
M300 S1396 P103
M300 S1318 P103
M300 S1396 P413
M300 S1174 P310
M300 S1046 P103
M300 S1174 P206
M300 S1567 P206
M300 S1567 P206
M300 S1396 P206
M300 S1318 P206
M300 S1567 P206
M300 S1567 P206
M300 S1396 P103
M300 S1318 P103
M300 S1396 P413
M300 S1174 P413
M300 S1318 P206
M300 S1046 P206
M300 S1174 P827
M300 S0 P206
```
  🔽  
```
;Scooter with Dwell
M300 S1567 P206
G4 P206
M300 S1567 P206
G4 P206
M300 S1396 P103
G4 P103
M300 S1318 P103
G4 P103
M300 S1396 P413
G4 P413
M300 S1174 P310
G4 P310
M300 S1046 P103
G4 P103
M300 S1174 P206
G4 P206
M300 S1567 P206
G4 P206
M300 S1567 P206
G4 P206
M300 S1396 P206
G4 P206
M300 S1318 P206
G4 P206
M300 S1567 P206
G4 P206
M300 S1567 P206
G4 P206
M300 S1396 P103
G4 P103
M300 S1318 P103
G4 P103
M300 S1396 P413
G4 P413
M300 S1174 P413
G4 P413
M300 S1318 P206
G4 P206
M300 S1046 P206
G4 P206
M300 S1174 P827
G4 P827
M300 S0 P206
G4 P206
```

Other than this, it seems the code `M300` actually works and can be used to play a sound at the end of the print - topic of this issue.

Michele Moramarco
Prusa Research

foorschtbar, 2024-02-27:
> It seems to work to me, also on MK4.
> 
> What seems to work best though is "duplicating" the duration with a `G4` "[Dwell](https://help.prusa3d.com/article/prusa-firmware-specific-g-code-commands_112173)" code - or even extending it by several milliseconds to generate pauses.

Yes, with the delays it works, but this is only a **workaround**. There is a PR with a fix waiting #3686 for approval...

closed 2024-04-11 by Prusa-Support