prusa3d/Prusa-Firmware-Buddy #1076: [FEATURE REQUEST] Treat USB media as SD card for serial control

compmstr, 2020-11-21:
It would be nice to be able to use the USB media plugged in to the Prusa Mini as an SD card for use with Octoprint.  If nothing else, it seems like it might alleviate issues with https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/739, since you would be able to remotely initiate a USB print, which does appear to properly pause with the filament sensor.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
Printer type - [MINI]
Optional upgrades - [Filament Runout Sensor]

**Is your feature request related to an existing problem? Please describe.**
I wanted to use up the last of some filament, and so I kicked off a print from OctoPrint.  The sensor never triggered the print to pause.  Luckily I was watching it, so I was able to manually pause and change the filament.  Researching the issue, it seems like printing from the USB is the only way to currently have the filament sensor properly work.
I then tried to use the 'SD' commands on octoprint (M20/M21, etc.), and they all appear to do nothing.  I'm assuming this is because the hardware doesn't have an SD card.  If the USB media storage could be treated like an SD card, this would allow us to remotely kick off and control USB prints from octoprint.

**Describe the expected functionality**
Have the SD card GCODE commands (M20, M21, M24, M25, etc.) use the USB media storage in place of a physical SD card slot.

JohnnyDeer, 2020-11-30:
Hi @compmstr fixing this commands is on our to do list.

pickworthi, 2021-12-22:
I would like to add another use case, given that the original reason for this issue has potentially been solved by later firmware releases.
The use case I am interested in is to use the M20 to M32 commands over the serial link to load/delete files on the Mini+ USB stick. This allows a local operator to fetch gcode files from a remote source, without having to physically walk there and back with the USB stick in his/her hand.

bschollnick, 2021-12-22:
> I would like to add another use case, given that the original reason for this issue has potentially been solved by later firmware releases.
> The use case I am interested in is to use the M20 to M32 [...]

I think this is an interesting idea.  But, I’m a bit concerned security wise, that this would allow any unvetted gcode file to erase files (or download files) to my system.  

Just the loading by itself, could potentially open the system up to bad actors (eg Malware, Virii, etc).

pickworthi, 2021-12-22:
The codes already exist in Marlin, and are implemented in other printers than the Mini+. I believe that the I3 supports them for example.
A bad actor would need to have access to the serial port of the printer, or the external controller (e.g. Octoprint). Both attack vectors are easily mitigated (e.g. lock on workshop door, isolated LAN sub net for Octoprint and firewalls).

Rose-David, 2022-01-18:
Plus, if there is a security concern, just have it default to OFF. I feel like my octoprint is secure, and would love these useful features, but I also totally understand not wanting the risk.

arekm, 2023-04-19:
M20 works according to paste here https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2387#issue-1336615121
(well, partially, no support for "L", not support for "T" option - both supported in MK3, none in MK4).

M23 unfortunately is failing.

Marlin 2 (which Firmware-Buddy is based; but based on quite old version) generally supports all (or most) SD commands *also* with USB medium, so there is a chance to get that working on Firmware-Buddy, too.

Prusa-Support, 2023-06-06:
Thanks for reporting.
The issue remains open.
However, on a side note, the issue starts looking obsolete now that Prusa Link and Prusa Connect can allow uploading and starting prints from remote - Prusa Connect is enabled for MK4 already and will be enabled for MINI in the near future.

Michele Moramarco
Prusa Research

MichaelWOlson, 2023-06-12:
> However, on a side note, the issue starts looking obsolete now that Prusa Link and Prusa Connect can allow uploading and starting prints from remote - Prusa Connect is enabled for MK4 already and will be enabled for MINI in the near future.

I'm not sure how wide spread it is, but for many of us that's not functional yet, it'll be nice when it arrives:

#3049

Prusa-Support, 2023-07-04:
Some work on transmission speed has already been done but issue #3049 remains open as we keep working on it.
Improving transmission speed is among our main objectives at this early stage of the Prusa Link/Connect wide adoption. 

Michele Moramarco
Prusa Research

jagerkin, 2024-06-14:
Comment to keep from auto-closing.

danopernis, 2024-09-02:
This one is (slowly) being worked on.

aymericblok, 2025-02-08:
I love the prusa connect, but i want to print without using the USB stick, because my roommate puts his prints on the USB stick, and when i use prusa connect and just upload my files, they go to his USB stick.

So i installed Octoprint on my 2nd phone, so i can print off of that, and he off his USB drive.

Ofcourse it will be possible to remove his usb every time i want to print, but i often forget that. Because i starts prints from another room