prusa3d/Prusa-Firmware-Buddy #1106: SW 4.2.1 Display task issue, blue screen.

marcingajewski, 2020-11-30:
I bought the reprap printer with bigger screen. 
I updated the printer's SW to latest version of 4.2.1 and it goes into blue screen when want to return from system info.
[image]
[image]
[image]

JohnnyDeer, 2020-12-01:
Hi @marcingajewski what exact type of printer you are using? Did you just flash the printer and checked version info followed with blue screen or you did something before? Thank you for more info.

marcingajewski, 2020-12-08:
Hi Johny,

I bought this printer as reprap from local man in Poland. He claims that only difference is bigger screen, I don’t have MPN unfortunately.
After a flashing to newer software (old was working fine) and getting to check the version it does it every time. The printer can work, print, do all stuff, but always when getting there and exit (sometimes it no need to exit, doing it after a while)
, blue screen appears. 

[image]

JohnnyDeer, 2021-01-18:
@marcingajewski thank you for information. Can you please try it on 4.3.0-RC1? Thank you

JohnnyDeer, 2021-08-02:
@marcingajewski could you please test it on latest FW version?

marcingajewski, 2021-08-02:
Sorted on 4.3.1. All works, thanks for sorting it

closed 2021-08-02 by marcingajewski