prusa3d/Prusa-Firmware-Buddy #1119: [FEATURE REQUEST] Heatbreak fan maximum PWM

jack-aaron-cooper, 2020-12-03:
**Printer type** - MINI

**Is your feature request related to an existing problem? Please describe.**
  As a user, I should be able to set the maximum PWM value of the heatbreak fan, so that if I want to use a filament that requires higher temperatures, I will not need to worry as much about filament jams. The current maximum PWM is set to 50%.

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/95447c177e0d3301713266200ed2620fe10551ae/src/common/config_a3ides2209_02.h#L83

**Describe the expected functionality**
  Have a section in the settings to change the maximum heatbreak fan PWM.

blerrgh, 2020-12-03:
see #997

JohnnyDeer, 2021-02-17:
I´m closing this issue as duplicate of #997.

closed 2021-02-17 by JohnnyDeer