prusa3d/Prusa-Firmware-Buddy #1131: [BUG] 4.2.1 + SuperPINDA - MBL is not compensating fully

vintagepc, 2020-12-06:
At the request in #636 I am starting a new issue for this. 

**Printer type** - [MINI]

**Printer firmware version**-  4.2.1

**Original or Custom firmware** Stock

**Optional upgrades** - Filament sensor

**USB drive or USB/Octoprint**
USB/Pronterface

**Describe the bug**
Mesh bed levelling does not appear to be fully compensating for the bed level. The amount of "squish" differs noticably from the left side of the bed to the right side. 
As noted in the linked issue, I have checked that the X rods are coplanar with the bed, and I do have a superpinda (so this issue should now be decoupled from any temperature effects.)

Bilinear levelling grid reported for this print is as follows:
Bilinear Leveling Grid:
      0      1      2      3
 0 +0.034 +0.061 +0.058 -0.045
 1 -0.058 +0.166 +0.137 +0.056
 2 -0.042 +0.089 +0.153 +0.167
 3 -0.119 +0.066 +0.201 +0.256

Notice the difference in line widths for 2 extrusions on the left and right side of the plate:
[image]
[image]

as well as the "plowing" on the left (and relatively smooth surface on the right) Black marker added for clarity/texture help.

[image]
[image]
[image]
[image]
[image]
[image]

**How to reproduce**
  Please describe steps to reproduce the behavior.

**Expected behavior**
Layer thickness should be consistent from left to right side of the bed. 

**G-code**
[3x3-calibration-prusa-mini_02mm_pla_mini_18m.gcode.tar.gz](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5649660/3x3-calibration-prusa-mini_02mm_pla_mini_18m.gcode.tar.gz)

**Crash dump file**
N/A

murk-sy, 2020-12-17:
Can confirm the same issue in 4.3.0-RC1 on 2 MINI+ printers. For the record, I have not touched print sheet profiles yet, assuming that matters.

Gaugeforever, 2021-06-01:
I just want to say I have this issue as well.  I have a brand new mini+ with the super pinda.  I am by no means an expert on 3D printing, but I'm also not new to it.  After attempting to do the calibration print multiple times, I can say without hesitation something is wrong.  Mine will drag in certain places on the sheet, in particular the right side.  I'm going to try heat soaking the entire thing and running the calibration then and see if I can get consistent results.  It is extremely frustrating to either be too high, or dragging the head.  Unfortunately, to make the middle of the sheet look good, it requires dragging, if I raise the nozzle on live z adjust so it doesn't drag, the middle of the sheet first layer looks awful.

I thought this issue was supposed to be fixed with the super pinda?

vintagepc, 2021-06-01:
The SuperPINDA just fixes the temperature variability which confounds the problem. 

This seems to be a separate issue and from  my observation, Z is moving very little as the head moves from left to right side of the bed, as though it's simply just not compensating enough for the measured distance.  I have also tried following the guide on https://help.prusa3d.com/en/article/xz-axis-skew-correction-mini_158518/ with no success - but it's worth noting that despite having raised the right end of the arm by more than a mm, there is still a very similar level of "squish" to the parts on the right end of the bed. 

If this was just arm sag then in theory that should be compensated for by the PINDA as it should also read "closer" and therefore raise Z as the arm tilts more.

Gaugeforever, 2021-06-01:
What's odd to me is that I can do the calibration Z print, get it leveled, then do the same print keeping what I entered last time, and it forces me to raise is by .2mm, if I repeat the same process, I need to raise it .2mm again.  It is the strangest thing.  Perhaps I should contact support.  I also noticed that either the sheet isn't flat at the bottom left or the magnets aren't strong enough or the heated bed is not flat because the bottom left has a tiny gap and try as I might I can't seem to get rid of it.

Gaugeforever, 2021-06-02:
I wanted to add a 9 point leveling print as proof that mine fails consistently front to back.  As there is no manual bed leveling option, nor is there an option to fix the broken calibration in the settings like other Prusa printers, I am basically stuck with a broken printer.  I will contact support tomorrow and see how refunds work.

[https://forum.prusaprinters.org/wp-content/uploads/2021/06/IMG_1998-600x500.jpg](Leveling Print)

Gaugeforever, 2021-06-02:
I've actually tried all of that.  I thought loosing the back of the bed would raise it.  None of it works.  Also, while I did do the xz askew test and it came out fine, that makes sense seeing as this is a yz askew problem, not xz.  They need to allow us to calibrate the bed leveling like in their other printers.  There simply is not a solution for y askew at this point.  I'm not sure if the steel sheet is not level, the heating bed is not level, the extrusions, I have no idea.  Whatever the case is though, the pinda probe is not compensating for it.  I can't go lower because if you look at the front, it's hard to see, the nozzle would drag and not print any lower.  I got it as low as possible without dragging in the front in this photo.

I'm a software engineer so I may branch the firmware myself and see if I can adjust the calibration there.  For now I'm going to add paper towel strips in the back, between the heating bed and the steel sheet to even see if correction of the bed would solve the issue.  If it would then editing the firmware manually (assuming it's open source) will allow me to fix this myself.

widgetlords, 2021-07-13:
@Gaugeforever We just struggled with the exact same issue for quite a while. In our case the heatbed wires were in reverse polarity. Swapping them around made the printer work correctly.

Gaugeforever, 2021-07-13:
I'll check the polarity of the wires.  However, I did manage to get this working correctly anyways with silicone bumpers around the bed screws so I could manually level the bed.  It works fine now.  I'll need to come up with a more permanent solution though.  Something prusa should supply if their manufacturing and return processes are not good enough (which they aren't).

vintagepc, 2021-07-13:
Pretty sure that's a red herring. Bed polarity shouldn't have any impact on MBL unless you disassembled something to replace the wire and "fixed" it during reassembly

widgetlords, 2021-07-13:
@vintagepc I thought the same but I have 3 of these printers, two had wrong polarity and had the exact same issue. I also disassembled and reseated the heat bed many many times with no change. The printer was mechanically perfect (measured with a level at 6 points and with precision calipers) but the bed levelling went completely wrong every time. The only reason I figured it out it was by swapping heat beds. The issue followed the heat bed.

vintagepc, 2021-07-13:
Interesting - did the bed LED also not work on those beds until you swapped the leads?

widgetlords, 2021-07-13:
Yes. I had thought they were damaged and the bed seemed to be heating correctly so I dismissed it out of hand.

vintagepc, 2021-07-13:
Definitely not my issue then, LED works as expected :)

widgetlords, 2021-07-13:
In that case, try firmware version 4.1.0. I found it fixed most of the issues before I found the bed problem.

Gaugeforever, 2021-07-13:
Well, just think about it like this.  Let's assume we are not crazy, and that Prusa is no so perfect that no bed will be more than .3 mm off front to rear.  There is no way to fix this without hacking away at your printer like I had to.  Even now, with my mesh readout within .1 mm across the entire bed, the mesh level system doesn't do a perfect job.  Part of me is convinced it doesn't actually do anything at all except read the thing in the firmware.

They need to A) Allow offsets to be input into the firmware like with the mk3.  B) Engineer a bed system creality has long ago solved by allowing the user to manually bed level.  C) Have a fucking return process that doesn't require video transfers for proof to support, tons of images of proof of shipping materials, EXACT SHIPPING PROCESSES, and charge the person for the shipping.  It was my understanding the last one was one of the reasons we pay Prusa more, when in fact it is worse than other companies.  It's not 2016 anymore.

Oh, any my micro usb broke off because of course it did.  So I hope my custom bed level holds, if it doesn't I'll have to attempt the world's worst solder or buy a new buddy board.  I will say, when it does work, after all this work, it is a fun little printer.  Not even close to worth the price, but fun nonetheless.

vintagepc, 2021-07-13:
Eh, I think I just need to get my rear in gear and implement MBL mesh specification in Mini404. Then I can feed in arbitrary mesh inputs and observe the deviance from "correct" as determined by the printer. Hopefully more effective than futzing around with random stabs in the dark ;)

vintagepc, 2021-07-14:
I'm curious for the other folks here with this issue - if anyone has a problematic bed and access to some sort of digital measurement probe/micrometer... Would you have time to find a way to mount it next to the PINDA to get "real" bed variance measurements at the 16 probe points? I added a simulated PINDA to the hardware sim and it'd be interesting to see if "real world" data that has issues translates into simulated compensation issues too. It'd be interesting to get a variety of data sets here to confirm or deny whether this is a firmware MBL bug. 

Note I'm not sure how valuable the pinda measurements as reported over serial are at this time; I haven't looked whether those are "raw' or modified/filtered in some way - hence the request for independent data from an external device.

Gaugeforever, 2021-07-14:
To be clear, my assertion isn't that the new PINDA probe is not accurate.  I believe it to be very accurate.  My assertion is that the firmware isn't doing anything with that data.  Or that if it is, it's not doing anything very useful.  I didn't have hyper accurate measurements, but I did use digital calipers with support and the pinda probe readouts seemed to be right on what I was measuring.  Having said that, it would be very hard for me to know within .1 mm if the pinda was accurate or not, and that would result in some of the issues I'm seeing if it wasn't.  Basically if the inaccuracy was on a very small scale, as opposed to the nearly full millimeter I started at.  

I plan on redoing my bed leveling system now that I know it works, so I could do this if you can tell me of a tool that can do this to the degree you desire.

vintagepc, 2021-07-14:
Sure - I'm not saying it is variability in the measurements either - but I'd just like to eliminate it as a variable in the early stages of investigating - at least, without confirmation they agree like what you've already done. If you've observed the measurements as reported by the mini FW over serial agree with your manual measurements then those values will be just as useful to start.

Gaugeforever, 2021-07-14:
Sure, here you go :)

[image]

npike, 2022-02-15:
Any updates on this?

Gaugeforever, 2022-02-16:
> Any updates on this?

Hello.  I don't think Prusa cares to be honest.  I've always felt that the biggest problem with Prusa printers is that they have an engineer in charge rather than normal business setup.  This has pros and cons, but one of the biggest cons is that engineers are never wrong.  Prusa seems to think their engineering is so great that this issue is essentially "fake news".  You should have seen my several hours with support.  They really do treat this like it's the first they are hearing about it.  If you want this fixed, you have to fix it yourself.

I ended up following a guide on the forums to the T and it has worked amazing.  I have a very flat bed and it prints wonderfully now.... after I have done extensive modding to it.  Here is the link to that post, you can see my results in my reply there [https://forum.prusaprinters.org/forum/user-mods-octoprint-enclosures-nozzles/prusa-mini-silicone-bed-leveling-mod/paged/15/#post-518551](link).  I hope this works for you as well because other than doing this mod I don't think there is a solution and I won't expect Prusa to acknowledge manufacturing variances on a scale that would let them admit this exists.

CarlosGS, 2022-06-29:
Here is a bug that could be causing this: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1352
Currently during mesh calibration the motor "stall" endstops remain enabled, so any small friction can cause the motion to stop prematurely during the grid. The problem is that this is currently ignored by the software, so grid calibration silently continues at bad locations, causing incorrect calibration.

You can replicate this problem by softly bumping the X or Y axes during bed leveling, while they are moving between probing points. The printer detects a collision and incorrectly resets that axis to 0, offsetting the entire grid.

PS: I also agree with @widgetlords that another cause could be an incorrect bed polarity, because the mosfet switching is done on the low side, and depending on the polarity the entire heat bed can PWM oscillate between 0 and 24V.

Prusa-Support, 2022-07-26:
Thank you for all of your feedback.

I assume that 4x4 mesh is not dense enough to outline very irregular surfaces with accuracy.
The calibration values are probed on 16 points and interpolated for the rest of the print area.
I guess the zones near the edge are trickier to define the bed mesh because there is no other calibration value to be interpolated beyond the edge. Also, the heat dissipation is supposedly stronger near the edge, which causes a temperature gradient that won't help a very uniform first layer.

The flatter the heatbed, the more accurate the interpolation between the 16 probed values.
This is most often a matter of assembly.
Some microscopic inconsistencies across the bed may turn more noticeable at very high temperatures, therefore causing minor alterations of the first layer quality at different temperatures.

For better adhesion on large prints, you can increase the bed temperature to compensate for the stronger heat dissipation towards the edges.
Another neat trick is lowering the first layer speed.

For help reviewing the printer's geometry and evaluating the heatbed uniformity, please reach our Customer Support.
https://help.prusa3d.com/article/customer-support_2287

The above could be enough to be able to safely print in most cases.
This doesn't mean the firmware can't be improved.

.
The possible bugs mentioned by @widgetlords and @CarlosGS are surely interesting. I guess these odd behaviors can be avoided if the heatbed is connected and assembled accurately, and if XYZ movements are smooth, but these two cases may deserve further study.
The reproducibility of such issues on multiple printers could help confirm bugs.

Michele Moramarco
Prusa Research

closed 2024-07-04 by github-actions[bot]