prusa3d/Prusa-Firmware-Buddy #1144: [QUESTION]<Prusa mini Fan Test 4.3.0>

Lino77, 2020-12-10:
Hi folks, I have a problem here and can't find a solution.
I have 3 Original Prusa (Delta) MINI Hotend fans and 3 original Prusa mini Print fans
on which I run the firmware 4.3.0. on an original Prusa mini yes I broke the seal and built a firmware myself and adjusted the steps and direction for the Bondtech.
of the 3 Delta fans one reaches
8105 RPM, according to the data sheet it should be 8000 RPM.
the second reaches 7666 RPM
and the third 7271RPM the first passes the fan test the second and third fails. would say that two out of three are broken or,
the same for the print fans, the first reaches 4696 RPM and passes the test, the second 3343 RPM and the third 3107 RPM and does not pass the test. I would say two of the three fans are also defective. what do you think

DRracer, 2021-01-11:
@Lino77 could you please post the logs from USB flash for each of the fans? Would be interesting to see how they behave.

Lino77, 2021-01-13:
SELFTEST START
printer serial:

Fan0 Started                    in diesem Bereich
Fan1 Started                    solten die werte liegen
Fan0 at 20% PWM = 454 RPM       150 -1950
Fan1 at 20% PWM = 2090 RPM      2350 -3750
Fan0 at 40% PWM = 1265 RPM      1250 -3950
Fan1 at 40% PWM = 4419 RPM      4750 - 5850
Fan0 at 60% PWM = 1979 RPM      2350 - 5050
Fan1 at 60% PWM = 5775 RPM      5950 - 7050
Fan0 at 80% PWM = 2577 RPM      3250 - 5950
Fan1 at 80% PWM = 6480 RPM      6850 - 8050
Fan0 at 100% PWM = 3107 RPM     3850 - 6650

Fan0 Started                    in diesem Bereich
Fan1 Started                    solten die werte liegen
Fan0 at 20% PWM = 819 RPM       150 -1950
Fan1 at 20% PWM = 2990 RPM      2350 -3750
Fan0 at 40% PWM = 2300 RPM      1250 -3950
Fan1 at 40% PWM = 5161 RPM      4750 - 5850
Fan0 at 60% PWM = 3397 RPM      2350 - 5050
Fan1 at 60% PWM = 6335 RPM      5950 - 7050
Fan0 at 80% PWM = 4176 RPM      3250 - 5950
Fan1 at 80% PWM = 7244 RPM      6850 - 8050
Fan0 at 100% PWM = 4696 RPM     3850 - 6650
Fan1 at 100% PWM = 8105 RPM     7650 - 8950
Fan0 Passed
Fan1 Passed

Fan0 Started                    in diesem Bereich
Fan1 Started                    solten die werte liegen
Fan0 at 20% PWM = 389 RPM       150 -1950
Fan1 at 20% PWM = 2352 RPM      2350 -3750
Fan0 at 40% PWM = 1097 RPM      1250 -3950
Fan1 at 40% PWM = 4577 RPM      4750 - 5850
Fan0 at 60% PWM = 1876 RPM      2350 - 5050
Fan1 at 60% PWM = 5959 RPM      5950 - 7050
Fan0 at 80% PWM = 2612 RPM      3250 - 5950
Fan1 at 80% PWM = 6801 RPM      6850 - 8050
Fan0 at 100% PWM = 3343 RPM     3850 - 6650  sollte der nicht mindesten 5000 RPM unter vollast haben 
Fan1 at 100% PWM = 7666 RPM     7650 - 8950  sollte der nicht mindesten 8000 RPM unter vollast haben lauf datenblatt von Delta 
Fan0 Passed
Fan1 Passed

closed 2024-07-04 by github-actions[bot]