prusa3d/Prusa-Firmware-Buddy #1149: [BUG]<Enter comprehensive title>

andy9a9, 2020-12-11:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI

**Printer firmware version**-  4.2.1

**Original or Custom firmware** Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint** - USB drive

**Describe the bug**
After device start-up, manual axis movement (over settings) have always the same values.
```
X=180
Y=-3
Z=0
E=0
```
If you manually (without power) move the X axis to 180, then move it over settings into middle of plate (lets say at 100). After power-off/on, it will have 180 value back even if it was moved to 100.

**How to reproduce**
Just restart the device.

**Expected behavior**
Automatically save previous XYZ positions from previous session.

bmangodt, 2020-12-11:
I think this is normal behaviour? After power down the printer has no idea what you do with the axis, or even with power with disabled steppers, so it will assume those standard position values as safe. Only after homing the printer, it has recalibrated itself and has an idea of the position of the axes and the values will match.

andy9a9, 2020-12-11:
Fine, after "auto calibration" before each manual moving it works.

I was thinking, that it will store the previous values while printer was not moved.

closed 2020-12-11 by andy9a9