prusa3d/Prusa-Firmware-Buddy #1151: [BUG] 4.3.0-RC1 Self Test Multiple Errors Crash

nskrotzki, 2020-12-11:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version**-  4.3.0-RC1

**Original or Custom firmware** Original

**Optional upgrades** - None

**USB drive or USB/Octoprint**
USB Drive

**Describe the bug**
My machine has never passed a self-test ever. I've brought it up with support and they can't identify the issue but when I do a manual self-test, essentially all the steps done during a self-test, the machine passes. Tried a self-test using the new test build to see if it would finally pass the test with the updated firmware and immediately get a USBH Thread Multiple Errors.

**How to reproduce**
I plugged in the USB Drive to try and capture the self-test crash logs and ran the test. I got through the fan test initially then it crashed, boot looped, and threw up the USBH Thread Multiple Errors. Trying to repeat the test after pushing the reset and trying the self-test immediately caused a boot loop and the USBH Thread Multiple Error. If I remove the USB and try the test again it works just fine without the drive. 

**Expected behavior**
Expected the Self Test to proceed as normal. 

**Crash dump file**
Error Image - [https://imgur.com/a/JqVQ25B](url)
Fan/XYZ Files Generated - [https://drive.google.com/file/d/1qELVmL98Gvo9ELqF16VnAWMxCulAliT2/view?usp=sharing](url)

**Video**
Error video - [https://drive.google.com/file/d/1uF8VvKCv6ZWO-GWsUUU8Uav9rCZLaUrz/view?usp=sharing](url)
I tested with a fresh flash and cleared USB. Tried the test with USB plugged in, and it'll crash. Then proceeded to do the test without the drive and it'll pass. Tested a 3rd time with a 'dirty' drive used in the first test and it'll crash.

JohnnyDeer, 2020-12-14:
Hi @nskrotzki thank you for reporting. So without usb plugged in is self test successful? Did you tried different usb stick? 
This function probably won't be in final.

AlexW-578, 2021-01-01:
Hi @nskrotzki and @JohnnyDeer
I had the same problem with the crash and BSOD, For me i passed fine after removing the USB stick.
Hope this helps

DRracer, 2021-02-08:
@AlexW-578 @nskrotzki there are still issues with the USB drivers, we are working on it with STM devs. Therefore the saving capability of test results has been disabled in FW 4.3.0-RC2 and final until the USB drivers are polished.

DRracer, 2021-02-08:
If you don't mind I'd close this issue, because it could have happened in 4.3.0-RC1 only.

closed 2021-02-08 by DRracer