prusa3d/Prusa-Firmware-Buddy #1153: [BUG]Self test resume variables are switched up

Fliens, 2020-12-14:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version**-  [4.3.0-RC1]

**Original or Custom firmware** [Original/Custom]

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
USB

**Describe the bug**
I ran the selftest and for the first time ever it failed (print fan)
But the thing that confused me even more was that during the test the print fan had the red x and on the resume the hotend fan had the red x so it seems like on the resume there are some variables switched!
[image]
[image]

Both photos were taken seconds after eachother

**How to reproduce**
  The printfan needs to fail (unplug)

**Expected behavior**
  The resume should show that the print fan failed

Lino77, 2020-12-14:
ich have the same and test 3 Hotend and 3 Print fan look @my post #1144 

I changed the fan test a bit so that the fan test runs to see where it is

test 1
SELFTEST START
printer serial: 

Fan0 Started                                   in diesem Bereich
Fan1 Started                                   solten die werte liegen
Fan0 at 20% PWM = 389 RPM       150 -1950
Fan1 at 20% PWM = 2352 RPM      2350 -3750
Fan0 at 40% PWM = 1097 RPM      1250 -3950
Fan1 at 40% PWM = 4577 RPM      4750 - 5850
Fan0 at 60% PWM = 1876 RPM      2350 - 5050
Fan1 at 60% PWM = 5959 RPM      5950 - 7050
Fan0 at 80% PWM = 2612 RPM      3250 - 5950
Fan1 at 80% PWM = 6801 RPM      6850 - 8050
Fan0 at 100% PWM = 3343 RPM     3850 - 6650  sollte der nicht mindesten 5000 RPM unter vollast haben 
Fan1 at 100% PWM = 7666 RPM     7650 - 8950  sollte der nicht mindesten 8000 RPM unter vollast haben lauf datenblatt von Delta 
Fan0 Passed
Fan1 Passed

test 2 

SELFTEST START
printer serial: 

Fan0 Started                                    in diesem Bereich
Fan1 Started                                    solten die werte liegen
Fan0 at 20% PWM = 454 RPM       150 -1950
Fan1 at 20% PWM = 2090 RPM      2350 -3750
Fan0 at 40% PWM = 1265 RPM      1250 -3950
Fan1 at 40% PWM = 4419 RPM      4750 - 5850
Fan0 at 60% PWM = 1979 RPM      2350 - 5050
Fan1 at 60% PWM = 5775 RPM      5950 - 7050
Fan0 at 80% PWM = 2577 RPM      3250 - 5950
Fan1 at 80% PWM = 6480 RPM      6850 - 8050
Fan0 at 100% PWM = 3107 RPM     3850 - 6650  sollte der nicht mindesten 5000 RPM unter vollast haben 
Fan1 at 100% PWM = 7271 RPM     7650 - 8950  sollte der nicht mindesten 8000 RPM unter vollast haben lauf datenblatt von Delta 
Fan0 Passed
Fan1 Passed

and 3
SELFTEST START
printer serial: 

Fan0 Started                                    in diesem Bereich
Fan1 Started                                    solten die werte liegen
Fan0 at 20% PWM = 819 RPM       150 -1950
Fan1 at 20% PWM = 2990 RPM      2350 -3750
Fan0 at 40% PWM = 2300 RPM      1250 -3950
Fan1 at 40% PWM = 5161 RPM      4750 - 5850
Fan0 at 60% PWM = 3397 RPM      2350 - 5050
Fan1 at 60% PWM = 6335 RPM      5950 - 7050
Fan0 at 80% PWM = 4176 RPM      3250 - 5950
Fan1 at 80% PWM = 7244 RPM      6850 - 8050
Fan0 at 100% PWM = 4696 RPM     3850 - 6650
Fan1 at 100% PWM = 8105 RPM     7650 - 8950
Fan0 Passed
Fan1 Passed

xthedakmanx, 2020-12-22:
+1, same issue. Never had the fans fail before and now they are

SELFTEST START
printer serial: CZPXSZ2020826PMNBCA

Fan0 Started
Fan1 Started
Fan0 at 20% PWM = 525 RPM
Fan1 at 20% PWM = 2243 RPM
Fan1 2243 RPM out of range (2350 - 3750)
Fan1 Failed
Fan0 at 40% PWM = 1622 RPM
Fan0 at 60% PWM = 2562 RPM
Fan0 at 80% PWM = 3369 RPM
Fan0 at 100% PWM = 4053 RPM
Fan0 Passed
SELFTEST END

JohnnyDeer, 2021-02-22:
@Fliens this issue should be already fixed, can you confirm that? Thank you

JohnnyDeer, 2021-05-12:
I´m closing this issue as solved.

PR #1230 (2021-01-13): fix switched fan results
Interchange fan0 vs. fan1 for selftest report.

Fixes #1153
closed 2021-05-12 by JohnnyDeer