prusa3d/Prusa-Firmware-Buddy #1154: [BUG]Wizard got stuck at preparing hotend

Fliens, 2020-12-14:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version**-  [e.g. 4.0.5, ...]
4.3.0-RC1

**Original or Custom firmware** [Original/Custom]
Original
**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
USB
**Describe the bug**
I just ran the wizard from the calibrate menu and when it tested the heating elements: hotend and heatbed it got stuck at preparing the hotend. The heatbed completet all it's tests but the hotend didn't even heat up!

**Expected behavior**
Hotend should usually heat up and complete the test

JohnnyDeer, 2021-01-11:
Hi @Fliens, could you please upload video of this issue? Thank you.

Fliens, 2021-01-13:
Unfortunately not, only had this issue once

JohnnyDeer, 2021-01-18:
We are unable to reproduce it or find out the problem. If you will have this bug again please open new issue with most information as will be possible. I´m closing this one as "can't reproduce". Thank you for understanding.

closed 2021-01-18 by JohnnyDeer