prusa3d/Prusa-Firmware-Buddy #1159: [BUG] Frequent random resets on FW 4.2.1

inferior-minion, 2020-12-15:
**Printer type** - MINI

**Printer firmware version**-  4.2.1

**Original or Custom firmware** Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint** - USB flash drive

**Describe the bug**
I've gotten at least FIVE resets on my Mini running 4.2.1. I updated as soon as it came out. I've attached the dump file from the latest one. Please help, this is very annoying and has been ongoing for a while.

Details of this particular reset:
I had printed something that took ~7 hours prior to starting this print which was a 12 hour print. It failed a few hours into the 12 hour print. The last thing I did on the printer was use the rotary encoder to select and start the print from the folder where the gcode was saved. Both prints were newly sliced with Prusa Slicer 2.3.0 Beta 2.
[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5692295/dump.zip)

**How to reproduce**
Start a print and wait.

**Expected behavior**
The print should have finished without a reset. There were no issues that I could see -- I was sitting right next to the printer the whole time during the failed print.

**Crash dump file**
Attached

JohnnyDeer, 2020-12-15:
Hi @inferior-minion, what is shown on display after reset? Please try new FW version 4.3.0-RC1.

inferior-minion, 2020-12-15:
I did just that - updated to the latest firmware v4.3.0-RC1 and I got a BSOD followed by a Watchdog Reset at the end of my **first print** on the new firmware. I sliced the file with the latest Prusa Slicer as well: v2.3.0 Beta3.

[gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5697154/gcode.zip)
[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5697160/dump.zip)

Here's a screenshot of the BSOD:

inferior-minion, 2020-12-17:
Hello @JohnnyDeer, it happened again. I got **another** print failure on 4.3.0-RC1:
Print was at 84%, then my MINI just froze for about 10 seconds mid-print. Next it reset and the attached Blue Screen appeared along with beeps. All I could do was reset, so I did and then saved the attached dump file. Again, the GCODE was sliced with the latest Prusa Slicer; 2.3.0 Beta3.
[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5710727/dump.zip)

**PLEASE HELP!** This is way too frustrating - I'm losing time, filament and patience!

JohnnyDeer, 2020-12-18:
@inferior-minion I printed your Gcode multiple times without problem. Could you please try print from different USB stick?

inferior-minion, 2020-12-19:
I've been using the 8gb USB drive that came with my Mini since day 1. I do have some extra USB drives I can test. I found a 16GB drive and did a full FAT32 format (not a quick format) on the drive in Windows 10. Then I copied over all the GCODE files from the original 8gb drive to the new test 16GB drive. I inserted the 16GB test drive (which works properly in Windows) into my Mini and got this screen:
[image]

vintagepc, 2020-12-29:
I can report a similar error on 4.3.0-RC1. 

Was also using the official mini USB drive. Reprinting with a different drive now.

[dump.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5751415/dump.bin.zip)

[image]

JohnnyDeer, 2021-02-22:
Hi @inferior-minion this BSOD was caused by Prusa Slicer 2.3.0 Beta 2, work fine with final. Error mounting USB should be fixed in 4.3.1. Did you tried update FW and use final version of Prusa Slicer?

JohnnyDeer, 2021-05-13:
@inferior-minion @vintagepc can you confirm if this issue is fixed by using 4.3.1 and final version of Prusa Slicer?

vintagepc, 2021-05-13:
I've not had this recur since using a different USB drive.

JohnnyDeer, 2021-06-07:
I´m closing this issue as stale and probably solved by new USB drivers on 4.3.2-RC1.

closed 2021-06-07 by JohnnyDeer