prusa3d/Prusa-Firmware-Buddy #1175: USB Disconnects after filament change

kghunt, 2020-12-17:
If you insert a USB while loading or unloading filament, once you have finished loading and return to the main screen the usual popup appears. If you dismiss the popup then the USB is shown and disconnected.

jkavalik, 2020-12-17:
It happened to me a couple of times, usually I just re-insert the usb but once I went to some other submenu (I think Filament, but not sure) and returned and the drive showed again without removing.

JohnnyDeer, 2021-01-12:
Hi  @kghunt @jkavalik, please try to use extension USB cable that might help.

jkavalik, 2021-01-12:
@JohnnyDeer I actually do use an extension cable already.

JohnnyDeer, 2021-01-18:
@jkavalik @kghunt we are working on find out and resolve this issue.

jkavalik, 2021-01-21:
@JohnnyDeer It just happened to me again so I can give a specific details:

Menu->Filament->Change filament
Remove usb drive and add new files into a directory, not root
Plug the usb drive back
Finish filament change (with one additional purge)
Exit the Filament menu
-> "newest file" print dialog is automatically opened but shows the newest file in root, not the one in dir so I cancel it
-> now the print menu item shows that no usb is connected
Preheat menu -> back
-> print menu item shows the drive is connected ad recognized, I can start the print from directory without problems (and without touching the usb drive)

I do mostly the same procedure quite often (putting new files onto the drive subdir during filament change) and most of the time the print item works without this problem so this is not a 100% way to trigger it.

JohnnyDeer, 2021-01-26:
@jkavalik thank you for details.

JohnnyDeer, 2021-05-06:
@jkavalik @kghunt does this issue happens to you on 4.3.2.-beta1? Should not because of updated USB driver.

JohnnyDeer, 2021-08-02:
I'm closing this issue for no responses.

closed 2021-08-02 by JohnnyDeer