prusa3d/Prusa-Firmware-Buddy #1176: [QUESTION]Community translations/languages

murk-sy, 2020-12-17:
Since there's UTF8 support for localizations, are there any plans to support even more languages, with translations provided by the community?

There are some considerations for this:

1. Translations can only be moderated by its contributors, the speakers themselves
2. Firmware space constraints may make it unrealistic to include every language
3. Compiling own firmware to support your own language requires breaking your warranty

Though perhaps not the most optimal, points 2 and 3 could be avoided by providing builds for languages in different regions - for example, 2 different builds would have English+European languages and English+Asian languages. Though to be frank I doubt point 2 is an issue that would have to be solved on the short term, but could be planned for.

DRracer, 2021-01-04:
@murk-sy good point, the answer is yes - we are planning to allow community translations in the future.

DRracer, 2021-07-30:
@murk-sy just FYI - PR #1522 is one of the building blocks for supporting community languages :wink:

closed 2024-06-07 by murk-sy