prusa3d/Prusa-Firmware-Buddy #1179: [BUG] Pausing during infill on 4.3.0 RC1

inferior-minion, 2020-12-18:
**Printer type** - MINI

**Printer firmware version**-  4.3.0 RC1

**Original or Custom firmware** Original

**Optional upgrades** - Filament Runout Sensor

**Printing from USB drive**

**Describe the bug**
When printing the infill of a print, the Mini pauses every few seconds for one second.

**How to reproduce**
Start a print with infill and wait for it to get to the infill part of the print.

**Expected behavior**
There should be no pausing when printing any part of the print.

**G-code**
I will attach the GCODE after the print finishes - it's still in progress.

**Video**
<link coming>

JohnnyDeer, 2021-01-12:
Hi @inferior-minion, please attach the Gcode, thank you.

DRracer, 2021-02-08:
@inferior-minion Do all your gyroid infill's (i.e. high data rate) cause this behavior? From the video it looks like the USB drive is not able to supply data fast enough (no idea so far what it may be the reason).

Could you please try another USB drive, disconnecting the ETH cable from your printer?

Could you please verify the same behavior on FW 4.3.0 final?

JohnnyDeer, 2021-05-13:
This issue will be closed soon because of no response.

closed 2021-05-17 by JohnnyDeer