prusa3d/Prusa-Firmware-Buddy #1182: [BUG] Z axis crash during XYZ calibration after X axis fail

murk-sy, 2020-12-18:
**Printer type** - MINI+

**Printer firmware version** -  4.3.0-RC1 

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**Describe the bug**
When running XYZ calibration, a failure on X axis that causes the carriage to stay at the leftmost side causes the entire gantry to crash without stopping on Z calibration.

**How to reproduce**

1. Start XYZ axis calibration (or wizard)
2. Printer homes
3. X axis is tested and **FAILS**. The carriage stays at the leftmost part of the X axis.
4. Y axis is tested.
5. Z axis is tested.
6. Old homing is ignored. The SuperPINDA is resting beyond the bed and thus does not detect nearing the bed *or* using the current position based on homing from point 2.
7. Entire gantry crashes into the bed without stopping, requiring a reset

**Expected behavior**
After X (or Y?) calibration fails:
- Ensure the ABL sensor is within the bed, or
- Ensure old homing values are used, or
- Prevent Z axis calibration from running at all

linx25, 2020-12-20:
Exact the same problem on my mini (without +) 
with the old firmware 4.2.1 the selftest work perfect

koenkooi, 2020-12-21:
I've had a virtually similar issue: the configured motor power wasn't high enough to move the X axis toward the base. If the gantry was completely to the left it would confuse stall for the endstop, which results in driving the extruder into the plate, hard. Obviously stallguard isn't working on Z.

I 'fixed' it by unplugging the printer and rapidly moving the X axis back and forth by hand. That seems to have loosened whatever seems to have been stuck. No problems since. The new X axis selftest in 4.3.0rc also shows no problems.

murk-sy, 2020-12-21:
> Exact the same problem on my mini (without +)
> with the old firmware 4.2.1 the selftest work perfect

Does that mean that even if x stalls on the left it doesn't crash? Or does it mean that x doesn't stall on the left at all? I would test but I really don't wanna destroy the printer further.

JohnnyDeer, 2021-01-18:
Hi @murk-sy, this issue will be fixed in 4.3.0 final. Thank you for reporting.

DRracer, 2021-02-08:
I'm sorry, this has been accidentally omitted from 4.3.0 final, but the good news is there is a fix already and it will be released in 4.3.1-RC1 shortly.

JohnnyDeer, 2021-02-10:
@murk-sy, 4.3.1 with the fix of this bug is out, please try it and update this issue, thank you.

murk-sy, 2021-02-10:
I'll try to recreate the X axis failure and also Y axis failure separately if possible, will let you know. Will also check #1268 while I'm at it, hopefully both by the end of the week.

murk-sy, 2021-02-12:
4.3.1-RC1 fixes this issue so I am closing it, thanks.

@JohnnyDeer I have done some testing on my machines and from what I can tell, Y axis failing *should not* cause a crash the same way, as there is a few mm left on the right side ears of the build plate before the ABL sensor stops detecting it when at X parked and Y -3 position.

**However**, since Mini Kit is now an option, I suggest adding the same failsafe to Y axis, to prevent a machine crash in case of incorrect assembly that would somehow leave more space than there should be.

Zynek90, 2022-05-25:
Hi, I am experiencing the same issue with firmware 4.3.4
X axis goes completely to the left and then Z doesn´t stop when lowering and requires reset.
I experienced this when calibrating for the first time but after a few tries it started to work. I printed three models all good. Then I turned printer off. Came back to it later  and now it cannot calibrate again. I don´t know what to do :(

Prusa-Support, 2022-05-27:
Hello @Zynek90, thank you for the report! From the description, it could be more a hardware issue, can you please send me a video of both the X going only to the left and the failed calibration from start to fail via email to me, info@prusa3d.com mentioning me in the subject? I'll review it from here and we'll find what caused the issue and find a solution for it. 

Alessandro Pantaleo
Prusa Research

closed 2022-05-27 by Prusa-Support