prusa3d/Prusa-Firmware-Buddy #1195: [BUG] Corrupted characters on Mini

chillygithub, 2020-12-28:
**Printer type** - MINI+

**Printer firmware version**-  4.3.0-RC1

**Original or Custom firmware** Original

**Optional upgrades** - SuperPINDA

**USB drive**

**Describe the bug**
  The last column of pixels in the letter "m" on the remaining time is missing.

**How to reproduce**
  Behaviour is seen on all prints.
  Mini is set to English language.

**Expected behaviour**
  The full letter "m" should be shown.

**G-code**
   Pre-sliced Benchy and Whistle supplied on USB stick with the Mini. 

**Picture**
[image]

JohnnyDeer, 2021-01-18:
Hi @chillygithub, we know about this issue and working on it. Thank you for reporting.

DRracer, 2021-02-08:
@chillygithub could you please verify the rendering is correct in FW 4.3.0?

chillygithub, 2021-02-08:
Thanks, this appears to be resolved.
I have only done shorts prints less than 1 hour. But on those the final line of the `m` is correct.

JohnnyDeer, 2021-02-08:
@chillygithub thank you for confirming, then can we close this issue as solved?

chillygithub, 2021-02-13:
Yes, this can be closed.

closed 2021-02-13 by chillygithub