prusa3d/Prusa-Firmware-Buddy #1196: [FEATURE REQUEST] Option to set Sheet Profile in Filament menu

chillygithub, 2020-12-28:
**Printer type** - MINI+
**Optional upgrades** - SuperPINDA

**Describe the expected functionality**
  Option to set the sheet profile from the "Filament" menu.

  Sheet profiles are typically change when changing filament / filament type.
  An option be included to change the Sheet Profile from the "Filament" menu, would save the need to go to the "Setting" menu.

AlexDCode, 2021-01-10:
I think this should be issued on the PrusaSlicer repository [here](https://github.com/prusa3d/PrusaSlicer)

Tomcus, 2024-06-13:
Closing this issue as duplicate: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3465

Please refer to the second issues with ideas and improvements in this area.

closed 2024-06-13 by Tomcus