prusa3d/Prusa-Firmware-Buddy #1198: [ENHANCEMENT] API with printer status

json81, 2020-12-29:
**Printer type**
MINI

**Is your suggested improvement related to an existing problem? Please describe.**
API is missing information when printer is in "pause" mode:
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1197

**Describe the expected functionality**
API is missing printer status. A status value should be added to show "Printing/Pause/Idle/Heating" for example.

chillygithub, 2020-12-29:
This would be helpful for an API based project I am currently working on.
Currently the only way to tel if the printer is printing or idle is to use the `project_name`.
A status field would be great.

der-pw, 2021-09-05:
Is there any news about this yet?
I would like to be notified when the M600 or M601 is running.
It would be very nice, to get a dataset for this in the telemetry data.

der-pw, 2021-12-26:
Maybe something will come soon. ;-)
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/lib/WUI/wui_REST_api.c#L124

closed 2024-06-13 by github-actions[bot]