prusa3d/Prusa-Firmware-Buddy #1200: [BUG]White Screen while printing

Fliens, 2020-12-29:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]
Mini

**Printer firmware version**-  [e.g. 4.0.5, ...]
4.3.0-RC1

**Original or Custom firmware** [Original/Custom]
Original

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
USB

**Describe the bug**
  So i started the print and when i came back to the printer to take a look at the print the screen was just white.
  Turning the knob didn't change anything but the print continues to run fine.

**How to reproduce**
  I mean I don't even know why it is there

**Expected behavior**
  Normal printing screen

[image]

JohnnyDeer, 2021-01-11:
Hi @Fliens, the white screen is caused by a poor connection on 
 LCD cable. Make sure it is connected properly.

Fliens, 2021-01-13:
I don't think it's the cable. When pressing the reset button (the one next to the knob) the screen works fine again

JohnnyDeer, 2021-03-01:
The only reason for white screen can be connection between LCD and Buddy board. Please check right assembly of cable and plugs. In case you will find some damage or wrong assembled plugs please contact our Tech Support, thank you.

michaelaswood1980, 2021-03-23:
I have also experienced the same issue whilst printing ABS in an enclosure.  I contacted PRUSA support and they have suggested that I update to the latest BETA firmware and try again.  I have checked and double checked the LCD ribbon and both ends are connected fine without any sign of damage.  I haven't as of yet tested the new firmware as I'm currently in the middle of a large print but my display turned white about 25 minutes into the print.  I will try to update this post once I've updated the firmware.

Fliens, 2021-03-23:
@michaelaswood1980 Try pressing the reset button when the print is finished
If the screen works fine after it, then it must be something in the software since you didn't change anything on the hardware

michaelaswood1980, 2021-03-23:
> @michaelaswood1980 Try pressing the reset button when the print is finished
> If the screen works fine after it, then it must be something in the software since you didn't change anything on the hardware

Yes, this is exactly what I have done on a couple of occasions. As you've mentioned it works fine after a reset but this is the 3rd time it's happened hence contacting PRUSA support. I'm hoping the BETA version will sort it.
I had wondered if it was the temperature of the enclosure but it's worked fine before and yours failed without an enclosure so must be a bug.

michaelaswood1980, 2021-03-24:
> Yes, this is exactly what I have done on a couple of occasions. As you've mentioned it works fine after a reset but this is the 3rd time it's happened hence contacting PRUSA support. I'm hoping the [...]

Just to update... I flashed the latest BETA firmware and touchwood it seems to have sorted the issue out. I'm in the middle of a second print and so far no white screen.  The update seems to have fixed some USB issue so possibly this was causing the screen to go white...

michaelaswood1980, 2021-03-28:
> Just to update... I flashed the latest BETA firmware and touchwood it seems to have sorted the issue out. I'm in the middle of a second print and so far no white screen. The update seems to have [...]

UPDATE - 28/03/21
This didn't solve the issue.  Still having the issue when printing, specifically in an exclosure.

[image]
[image]

kymlalu, 2021-04-06:
Try to plug printer into different wall socket. I had same issue  in one specific wall socket, after plugin it in different one it seems OK.

DRracer, 2021-05-06:
Everyone experiencing this issue - please note the root cause of this is disturbance in communication with the LCD. Usually it is caused by improperly stuck connectors into any of the boards (Buddy and/or LCD), especially if the LCD runs fine after restart of the MINI. It is not a firmware issue.
We have not seen so far any relation to ESD and/or power supply.

geeanas, 2021-09-06:
My screen went white after a short print and it hasn't go back. When I turn the printer on the screen looks normal until the rebooting loading bar finishes that it goes white. I've contacted prusa but they ran out of ideas on what could be the issue. So far I've updated the firmware, disconnected and connected all cables, check for continuity on the screen cable, and done every suggestion. If you have any idea, please let me know.

Please watch video for reference. https://youtu.be/rLRJHlLB-XQ (sorry about the background noise in the video)

michaelaswood1980, 2021-09-06:
Try formatting the USB pen or try using an alternative one and see if it
works.

FilipPyrek, 2021-11-03:
I have the same problem ✋  Restarting the printer always helps.
But input controls work, so I'm able to blindly change the filament etc.
Usually it's just white.. but from time to time it happens that it's rotated by 90° etc. As mentioned in the other posts above.
[image]

Starbender1031, 2021-11-13:
I have the exact same issue, except it only happens when I slice with Cura. PrusaSlicer works fine

AlessandroNava, 2021-11-17:
I had same issue, fw 4.3.3 and Alice with prusa slicer.
Temp 21/22 degree 
I Hope Will resolve....my first time with this problem!
The actual print ends in 8 hours. I hope that with the reset it resolves

AlessandroNava, 2021-12-15:
Solved....THE PROBLEM is USB
i chenged a USB PEB DRIVE

Prusa-Support, 2022-05-17:
Hello! Sorry for intervening so late on this thread. Is the issue still relevant? 
For both completely white screen and screen with random parts, rotated areas: as @DRracer mentioned, it's about interferences, very likely due to loose cable connection. Or could also be bad cable itself. 
But also bad USB, especially when it's completely white, bad USB in terms of corrupted, or perhaps not FAT32-formatted, as @AlessandroNava mentioned. 

I am closing the issue for now, in case any of you is still experiencing issues with this, please let me know via email info@prusa3d.com mentioning me ("attn: Alessandro"), we'll look into it. 

Alessandro Pantaleo
Prusa Research

closed 2022-05-17 by Prusa-Support