prusa3d/Prusa-Firmware-Buddy #1201: [QUESTION] Any official firmware "Todo List"?

padigree, 2020-12-30:
Hey, watching this firmware thread since creating. Often after a "feature request" I saw "it's already on our to do list". 

Is there something official? Maybe something we could see some progress? There are some relatively simple things (display timeout for example, request created 6 months or so ago) which I haven't seen any more - so maybe people could check what's the actual status of the future planned features - and can also see if there is something not that important for "soon to be out" updates. 

Anyway, thank you for your great work :) 
Kind regards

owenwastaken, 2021-01-06:
Yeah, it would be really nice to have a trello or something so we can see a roadmap.

blerrgh, 2021-01-07:
Seconded, this would be really nice to see.  As you say, there are many promised features like silent mode and power panic that have been in progress for over a year now, it would be nice for just a bit of communication as far as what is actually on the timeline.

riodoro1, 2021-01-19:
I would love to get WiFi and skew correction the most.
I'm guessing that the buddy board is set to become the next Prusa workhorse used in the whole lineup of FDM machines. The Mini then would be something of a development platform, but I have to say it seems like the work is very slow on it.
Hence why it would be great to finally know what can we reasonably expect from this machine and when.

Noxmiles, 2021-03-11:
Yes, would be perfect. I search in Google for "When get's the Mini WiFi Support", looking in forums, everywhere. A official ToDo List would calm down and prevent the same questions.

riodoro1, 2021-03-29:
Great communication from Prusa. This GitHub almost looks dead.
Cool machine but it could be made so much better in firmware.

padigree, 2021-04-07:
> Great communication from Prusa. This GitHub almost looks dead.
> MINI is a cool machine but it could be made so much better in firmware.

That's why I'm building a voron right now. After that, I'll maybe sell my mini+. Was waiting for the wifi feature since the release, nothing changed. Also the thing with esteps, which I have to use as custom gcode every time I load filament. Good Maschine but not really finished.

riodoro1, 2021-04-07:
> Also the thing with esteps

Oh yes. Prusa assumed their extruder is so great nobody would want to change it. They even disabled eeprom in marlin to make it harder to change anything. Quite a shitty attitude to take an open source firmware, disable a load of useful features users were familiar with and put a hw lock on custom firmware while not working on the official one.

d-zalewski, 2021-05-15:
> Oh yes. Prusa assumed their extruder is so great nobody would want to change it. They even disabled eeprom in marlin to make it harder to change anything. Quite a shitty attitude to take an open [...]

Prusa should buy Bondtech and include as an option to be fitted when ordering a printer. Including better support in the firmware.

danopernis, 2024-04-17:
No, there is no official _and_ public firmware TODO list.

closed 2024-04-17 by danopernis