prusa3d/Prusa-Firmware-Buddy #1207: [BUG] Wrong X value after startup

sfrayy, 2021-01-04:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version**-  4.3.0 RC1

**Original or Custom firmware** Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
USB

**Describe the bug**
After powering the printer and staring a print the Auto Bed Levelling starts as usual but the PINDA does not probe at the correct points in X direction. Therefore the first layer gets messed up. The "Test XYZ-Axis" Calibration or stopping and restarting the print fixes this problem. Once fixed, the bug does not reappear until the next restart.

**How to reproduce**
Turn off power. Turn on power. Print. 

**Expected behavior**
PINDA probes at the correct points. 

**Video**
https://youtu.be/0BhYDexB2rs

JohnnyDeer, 2021-01-12:
Hi @ sfrayy, check out this link and try the steps outlined there, thank you.

sfrayy, 2021-01-14:
Thank you for your help @JohnnyDeer. A simple firmware reset fixed it for now. I will keep an eye on it and see if it reappears.

closed 2021-01-14 by sfrayy

sfrayy, 2021-02-18:
UPDATE in case you come across the same issue:
The problem reoccured and i realized it appeared only when the printer was turned off for longer periods of time e.g. for a day. A restart after 5 minutes did not produce this bug. I followed the steps according to the linked post and ended up with customer support, who replaced my board. The issue did not occur with the new board, so I guess it was a hardware bug.