prusa3d/Prusa-Firmware-Buddy #1210: [ENHANCEMENT] Change default nozzle head axis position to be top left back corner?

AlvaroLuken, 2021-01-04:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

Enter what type of printer or upgrade the enhancement applies to.
**all printers

**Is your suggested improvement related to an existing problem? Please describe.**
After every print, at least for my MK3S, the nozzle places itself in like mid-center heatbed level - it makes it hard to clean the heat-bed + nozzle after each print as I have to waste time manually going in and moving the z-axis to >200 so that I can get a good angle for a nozzle clean and the bed is liberated from the nozzle being an obstruction for a wipe-down. If after every print (and really by default) the nozzle head would move to the upper top left corner, it would make it much more efficient to print and clean!

**Describe the expected functionality**
The default nozzle head position should be the upper top left corner for easier cleaning and printing.

ogland, 2021-01-12:
You can do this yourself very easily in your end gcode.

AlvaroLuken, 2021-01-13:
I am not an expert user, why not either make the default position easy to change from the menu controls?

closed 2024-06-17 by github-actions[bot]