prusa3d/Prusa-Firmware-Buddy #1218: Printing fails if speed is changed from the dial

cpinamtz, 2021-01-07:
**Printer type** - MINI

**Printer firmware version**-  4.2.1

**Original or Custom firmware** Original

**Optional upgrades** - No upgrades

**USB drive**

**Describe the bug**
Today I've discovered the reason why the [famous clicking](https://forum.prusaprinters.org/forum/?wpfs=prusa+mini+clicking) issue happens (at least with my MINI). As anyone who desires to get her creation as soon as possible, I've always been tempted to modify the original speed during the printing. Sometimes a bit (from 100 to 115) or even more (up to 180 for easy and low-poly designs). Then the clicking issue appeared. 

What I've discovered is that there is no problem if I don't modify the original printing speed for what the gcode was generated but even if I decide to **make it slower** then the clicking issue appears. That leads me to think that maybe there is some issue with the process related to recalculate the speed to pull the filament to the extruder in order to sync it correctly with the new speed.

[image]
In the attached image, you see clearly how the first layer is perfect but once the second one began, I speed up the printer a bit (from 100 to 115) and then the clicking issues appeared.

**How to reproduce**
Just print whatever and change the printing speed

**Expected behaviour**
Clicking issue

JohnnyDeer, 2021-01-18:
Hi @cpinamtz, we are unable to reproduce this problem. Did you tried it on 4.3.0-RC1? Can you provide Gcode and video? Thank you

JohnnyDeer, 2021-05-13:
This issue will be closed soon because of no response.

closed 2021-08-02 by JohnnyDeer