prusa3d/Prusa-Firmware-Buddy #1223: [ENHANCEMENT] Disable layer fun on filament change when its printing

daguilera-stc, 2021-01-08:
I did not find any related cases, I beg your pardon if I am in error.

**Printer type** - [MINI]

When you print a large piece up to near the lower left edge, if you do a filament change in the first layers, when the strusor is still close to the bed, the layer fan cools the bed and the piece peels off, especially if it takes longer make the change.

Maybe you could make sure the fan is off when the printer leaves the piece on the way to the corner for the filament change.

Thank you very much and greetings.

Prusa-Support, 2023-11-04:
Thanks for sharing.

I believe this behavior was inherited from "MK3's firmware" actually.
https://github.com/prusa3d/Prusa-Firmware/issues/3670

Our developers will look into this.

Michele Moramarco
Prusa Research

jvasileff, 2023-11-09:
> I believe this behavior was inherited from "MK3's firmware" actually.

Oh, maybe the fan configuration on the MK3 just made it less notable. Either way, thanks for putting this on the radar.

giacomosiena, 2024-04-15:
This is still an issue on MK$ 6.0.0-RC2

CZDanol, 2024-05-02:
Internal ticket: BFW-5474

Prusa-Support, 2025-12-06:
Closing as solved by FW 6.4.0 (final).
Thanks for your input.

Michele Moramarco
Prusa Research