prusa3d/Prusa-Firmware-Buddy #1225: [BUG] Filament Change Failure

FrenzyIncarnate, 2021-01-11:
[MINI]
4.2.1 Final
Bondtech Extruder
USB flash drive

**the bug**
I encounter a problem when I unload a higher temperature material and load a lower temperature material through printer's filament menu. When I unload PETG and load PLA, some amount of PETG remains in the nozzle. And since PLA's temperature is not enough to melt PETG, the extruder rams the PLA filament against solid PETG and this causes the olives at the extruder side to give way. This rendered my printer unusable until I replace the PTFE tube.

**How to reproduce**
  Use printer's filament change option and swap from PETG to PLA (probably few times)

**Expected behavior**
Hotend temp should be stayed at previous filament loading temp if its higher. Otherwise it shouldn't be a problem
**G-code**
 No G-code . its the firmware issue

**Crash dump file**
No dump file

**Video**
No video

JohnnyDeer, 2021-01-18:
Hi @FrenzyIncarnate, we are doing this kind of changes on daily basis without any problems. Looks more like you had bad piece of olive, or badly assembled one. Please try it with replaced one, video would be very helpful. Thank you

FrenzyIncarnate, 2021-01-18:
Hello @JohnnyDeer . This issue happened to me twice.  First one as you can guess with the factory assembled one. And I have used the spare olives and made one myself. After few months that also gave way .. I had to take another spares from Metatechtr.  I also know that some people use the last used material from load filament list to prevent this kind of issue.

FrenzyIncarnate, 2021-06-01:
[image]
Hello @JohnnyDeer I have uploaded it.

JohnnyDeer, 2021-09-06:
I have found that assembling of this part was changed together with a new molding tool, unfortunately bowden on your printer was assembled before this changes. So this really looks like faulty piece. If you will have another similar problem, do not hesitate to contact our Tech Support.

Prusa-Support, 2022-05-22:
Hello @FrenzyIncarnate! Was the issue resolved in the meantime through our Technical Support? In case there are more issues related with this, do not hesitate to let me know info@prusa3d.com mentioning me ("attn: Alessandro"), we'll find a solution for it. Closing the issue for now based on the info shared by @JohnnyDeer. 

Alessandro Pantaleo
Prusa Research

closed 2022-05-22 by Prusa-Support