prusa3d/Prusa-Firmware-Buddy #1227: [BUG] Crash while printing after 10-30min

nachtschatt3n, 2021-01-12:
**Printer type** - [MINI]

**Printer firmware version**-  [4.3.0 RC1 2691]

**Original or Custom firmware** [Original]

**Optional upgrades** - [none]

**Octoprint** - Version 1.5.2 - Version 0.17.0, running on Raspberry Pi 3 Model B Rev 1.2

**Describe the bug**
Crash and bluescreen, the printer is unresponsive and beeps. Check screenshot

**How to reproduce**
I send the gcode of my model to octoprint via PrusaSlicer 2.3.0+win64 and after 10-30min the printer crashes with a bluescreen. 

**G-code**
[assortment box 3x3 square_0.15mm_PLA_MINI_5h42m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5802076/assortment.box.3x3.square_0.15mm_PLA_MINI_5h42m.gcode.zip)

**Crash dump file**
[image]

nachtschatt3n, 2021-01-12:
I downgraded and tried again and now I get a MINITEMP error. Its a broken thermistor and the error message cloud be a little more descriptive

JohnnyDeer, 2021-01-18:
Hi @nachtschatt3n, thank you for reporting. We are working on this issue.

JohnnyDeer, 2021-02-22:
@nachtschatt3n could you please try test this issue on 4.3.1? Printing via USB should be fixed, if you are using OctoPrint there we still did not found reason. Thank you

nachtschatt3n, 2021-02-22:
@JohnnyDeer sorry I already replaced the thermistor and had no issues any more so far. Maybe someone else can try? It only appears with a broken bed thermistor

JohnnyDeer, 2021-02-22:
@nachtschatt3n that's ok, BSOD while printing via OctoPrint caused by broken thermistor cable is know bug. Then I will close this issue. Thank you for update.

closed 2021-02-22 by JohnnyDeer