prusa3d/Prusa-Firmware-Buddy #1231: [BUG] Prusa Mini crashes during first layer on 4.1.0, 4.2.1 and 4.3.0-RC1

d4rken, 2021-01-13:
**Printer type** - [MINI]

**Printer firmware version**
* Buddy-4.1.0-final+1923.bbf
* Buddy_4.2.1-final+2072.bbf
* Buddy_4.3.0-RC1+2691.bbf

**Original or Custom firmware**
[Original]

**Optional upgrades**
* Filament Sensor
* Bondtech Extruder

**USB drive or USB/Octoprint**
* Octoprint

**Describe the bug**
Printer crashes during first layer.
* 4.1.0 & 4.2.1 shows "Watchdog error"
* 4.3.0-RC1 shows a blue screen.

**How to reproduce**
Just by printing this file:
[OPM_Interlocking_Module2.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5811212/OPM_Interlocking_Module2.zip)

**Crash dump file**
[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5811278/dump.zip)

**Video**
https://www.youtube.com/watch?v=2d9lwOJqZrE

[image]

d4rken, 2021-01-13:
Crashes also when printing from USB.

JohnnyDeer, 2021-01-18:
Hi @d4rken, thank you for reporting. We are working on this issue.

d4rken, 2021-02-05:
Same with latest: Buddy_4.3.0+2756.bbf

d4rken, 2021-02-18:
This was likely caused by a faulty heat bed thermistor cable. I could confirm with a multimeter that the cable was faulty in a very specific position. After replacing it the errors went away.

I'm leaving this open as I only saw the display show an error relating to the thermistor in 1 out of 10 cases, so there may still be an issue with the error code?

JohnnyDeer, 2021-02-22:
@d4rken thank you for finding out and updating info. Yes, lets leave this issue open, BSOD should not happen. Can we close #1234 ? Seems to have same reason you find out. Thank you

d4rken, 2021-02-26:
I don't know, up to you. I'd say this can only be decided after fixing the BSOD here, and then comparing it with the BSOD in the other ticket. The memory/register dump within the BSOD  looks different though :thinking:

d4rken, 2021-09-08:
I'm closing this as I think this was a thermistor issue and according to the release notes, it was fixed in 

https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.3-RC

closed 2021-09-08 by d4rken