prusa3d/Prusa-Firmware-Buddy #1234: [BUG] Prusa Mini crashes during print on 4.3.0-RC1

d4rken, 2021-01-15:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type**
* [MINI]

**Printer firmware version**
* 4.3.0-RC1

**Original or Custom firmware**
* Original

**Optional upgrades**
* Filament Sensor
* Bondtech extruder

**USB drive or USB/Octoprint**
* Octoprint

**Describe the bug**
* Bluescreen

**How to reproduce**
* Just print

**Expected behavior**
* No crash :wink: 

**G-code**
[cherry_vista_badger_long_0.2mm_FLEX_MINI_1h14m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5821372/cherry_vista_badger_long_0.2mm_FLEX_MINI_1h14m.zip)

**Crash dump file**
[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5821392/dump.zip)

**Video**
* It just halts the printing and shows a blue screen

[image]

zoltan-l, 2021-01-25:
Hi d4rken,
can you please share info - did it happen at the start, during or ... which stage of print?

d4rken, 2021-01-25:
After 19m51s, on Layer 20/334.

d4rken, 2021-09-08:
Closing in favor of #1231

closed 2021-09-08 by d4rken