prusa3d/Prusa-Firmware-Buddy #1236: [QUESTION]<Enter comprehensive title>

padigree, 2021-01-17:
Please, before you create a new question, please make sure you searched in open and closed issues and couldn't find anything that matches.

Will there be a stealth Mode known from the mk3 in future firmware releases? 

Kind regards :)

Dev01-D, 2021-03-18:
I would love to see this integrated. I was toying with compiling my own marlin build however not keen on having to break the board in order to flash custom firmware.

DRKMSTR, 2021-07-24:
Bump

Any Dev Response?

Dev01-D, 2021-08-02:
I keep seeing these and see them being closed. do we have a current one?

JohnnyDeer, 2021-09-06:
Hi, unfortunately stealth mode is not on future feature list so this function will not be developed. Thank you for understanding.

plaetzchen, 2021-09-06:
Is there a reason why it won't be on the list?

abacabie, 2021-09-06:
> Is there a reason why it won't be on the list?

Probably because its pointless on the mini anyway, this thing is so quiet when set up correctly that the loudest part is the fans, stealth mode  is only noise reduction that you can do yourself with the slicer by slowing the print and acceleration, it offers nothing but drawbacks to mini, I'm glad they are not wasting time on this.

Dev01-D, 2021-09-06:
A little disappointed. I think it's a great feature. Having come from an Ender 3 with silent steppers to a Prusa Mini the mini is louder. The fans yes are one of the louder points but drivers could be a lot quieter. Stealth mode is not managed in slicer software. It is a feature in the firmware. If it were not important they wouldn't have it one the mk3.

plaetzchen, 2021-09-06:
While I appreciate that it’s not important for your use case it could be for mine. My printer runs in my study and I am working on my desk right next to it. Any dB I can kill is important. While the are a noise source they are constant and easy to “filter out”, the steppers can by noisy in some movements and super quite in others. That can be annoying. The possibility of adding stealth later was one of the reasons why I decided for the Mini.

DRracer, 2021-09-10:
Let me clarify the _Stealth mode_ in more detail. It is already implemented and in use. The MINI/MINI+ is equipped with the TMC2209 motor drivers, where after multiple tests we have enabled StealthChop by default for X, Y, and Z-axis. 
The name _Stealth mode_ comes from the MK3/S/+ and basically enables the same settings of its TMC drivers (StealthChop).

On the MINI+, the only exception is the extruder motor, where due to the bowden and some other reasons it performs better and reliably with the SpreadCycle enabled (that's the other mode of the TMC motor drivers).

Here is a snap of the code from the current MINI+ firmware: `include/marlin/Configuration_A3ides_2209_MINI_adv.h`
```
   /**
   * TMC2130, TMC2160, TMC2208, TMC5130 and TMC5160 only
   * Use Trinamic's ultra quiet stepping mode.
   * When disabled, Marlin will use spreadCycle stepping mode.
   */
    #define STEALTHCHOP_XY
    #define STEALTHCHOP_Z
//  #define STEALTHCHOP_E
```
Since the _Stealth mode_ is already used by the MINI+, we currently don't plan any further changes.

The other sources of noise are typically cooling fans. On the MINI+ have tested various scenarios with different materials. Based on the results, we have optimized first the firmware - lowering the speed of the extruder cooling fan to the lowest level while keeping the heartbreak safe. Next, we have tweaked the fan speeds in the PrusaSlicer print profiles for each material (i.e. not all profiles need the print turbine running at full speed).

Regarding the diagonal moves, we have found a way to improve them and we are currently testing the new approach. However, please give us more time to verify this change as it is touching the “hearth” of the print engine and requires extensive testing across the entire system.

Prusa-Support, 2022-05-22:
Hello! Sorry for intervening so late. I'll close the thread based on the info shared by @DRracer. 

Alessandro Pantaleo
Prusa Research

closed 2022-05-22 by Prusa-Support