prusa3d/Prusa-Firmware-Buddy #1239: G CODE / M CODE FOR BED LEVELING MINI

kakafon2205, 2021-01-18:
Hi I have a question whether it would not be possible to incorporate mesh bed leveling into the slicer or rather into the interface of a menu on the printer in the future. All the time before I catch the first layer, there are delays with cooling, measuring flatness and reheating. If it were possible to have it measured during the first print and after the detection of a poorly printed first layer, run the program behind this block. Or it would be possible to dig up your own program header in the slicer, but then be able to run the program from a certain block on the printer. This is how I have to modify the program for the first run and then have another one for the next iteration. 
printer type MINI

hasechris, 2021-02-08:
Hi @kakafon2205,

im using octoprint with this gcode. Please delete the comments if you want to use this.
```
M155 S0                     <-- disable temp reports
@BEDLEVELVISUALIZER    <--- hint for the Level Visualizer Plugin, that next output from Printer is MBL Table
G29 T                          <-- this triggers the MBL
M155 S1                     <-- enable temp reports every 1 seconds
```

You will get this output from "G29 T"
```
Recv:       0      1      2      3
Recv:  0 -0.192 -0.108 -0.111 -0.123
Recv:  1 -0.069 +0.084 +0.074 +0.057
Recv:  2 +0.017 +0.090 +0.098 +0.114
Recv:  3 -0.120 -0.039 +0.041 +0.096
```

Greetings
hasechris

hasechris, 2021-02-08:
Hi @kakafon2205,

soo, this was the background info. Back to the second part of your question. I think it is not a good practice to skip the MBL run infront of every print. This 30 Seconds will stop you from getting failed prints, which you cant see before 2-3 hours of printing...

Best Regards
hasechris

kakafon2205, 2021-02-08:
Yes. of course. You cnow beeter what is best from me :-) men.
But werry thanks for answer and i try learn programming .

Best regards.

JohnnyDeer, 2021-05-13:
Hi @kakafon2205, we don't consider this request is good idea, can cause more issues. You can skip MBL by custom Gcode as @hasechris wrote. I´m closing this issue. Thank you for understanding.

closed 2021-05-13 by JohnnyDeer