prusa3d/Prusa-Firmware-Buddy #1244: After filament unload, no prompt to ask if succesful

alfonsvh, 2021-01-19:
Up to v4.3.0-RC1, a prompt would come up after unloading or changing filament to ask whether the unloading was succesful. In RC2 this prompt does not appear.

JohnnyDeer, 2021-01-20:
Hi @alfonsvh, thank you for reporting.

adynis, 2021-01-24:
you convinced me to install RC2 , I didn't like that unloading prompt :-)

zoltan-l, 2021-01-26:
Open issue https://dev.prusa3d.com/browse/BFW-1702

DRracer, 2021-01-26:
@adynis I'm afraid the "unload successful" message is to be there from now on. It has its serious meaning once you run into unloading issues.

zoltan-l, 2021-01-26:
@alfonsvh 
double checked with developers and designers team. The function for 4.3 RC2 has been adjusted in the following way:
1.When you use menu FIlament->Unload filament , there is no need for this confirmation massage, if the filament unload did not succeed you are returned directly to Unload filamentmenu  item, so you just Unlod once again
2. More important - when you use during printout Tune-> Change filament command or you send M600 G_code, the printer  first stops, unloads the filament and asks a reasonable question "Was filament unload successful"? If not the printer allows you manually intervene and remove the filament before it starts with new filament load.

Closing the issue with parameter : works as designed

closed 2021-01-26 by zoltan-l

alfonsvh, 2021-01-26:
Okay, thanks a lot!