prusa3d/Prusa-Firmware-Buddy #1258: [FEATURE REQUEST] Set a default temperature for a filament on the printer

kurvalak, 2021-01-23:
**Printer type** - [MINI]
**Optional upgrades** - [e.g. Filament Runout Sensor]

**Is your feature request related to an existing problem? Please describe.**
When we load a filament, there is default temperature for the type of filament. More than often, those settings are not appropriate for some type of filaments. For some printers actions like Purging and First Layer Calibration, if the purging is not done at the correct temperature, it could leads to clogging.

The alternate method is to purge manually by setting a temperature and using the the extruder "Axis" in the menu. 

I'm aware that it is possible to change the temperature in the PrusaSlicer for a print and also that it is possible to change the temperature while a print is being done on the printer and those functionnalities do not cover this requirement.

**Describe the expected functionality**
As a user, I would like to be able to define a filament default temperature on the Prusa Mini Firmware for the following actions: Purging and First Layer Calibration

[Link to a forum post on the topic.](https://forum.prusaprinters.org/forum/general-discussion-announcements-and-releases/how-to-set-a-default-temperature-for-a-filament-on-the-printer-not-the-prusaslicer/)

XristophD, 2022-12-01:
I had a similar problem today with a wood-based-filament that doesn't fit any of the predefined temperature settings. 
Loading as PLA will heat up the nozzle too high, therefore First Layer Calibration is not possible or doesn't really make sense since the temperature settings are completely different. 

This feature request is open since 23 months, any update from the Prusa Team on this?

Prusa-Support, 2022-12-27:
Thank you for sharing your idea.

Setting custom pre-heat temperatures may not be necessary and may be unsafe for beginners.

Please notice that the first layer zig-zag calibration line doesn't need to be repeated for every single material. Such a simple g-code would work with a vast range of basic materials.
It should be set once, as an initial reference.
Further fine-tuning could be done "on the go" upon observation of a standard solid first layer (100% infill + perimeters) where the specific print temperature is set via the slicing program (e.g. Prusa Slicer) during the g-code generation.

The final temperature is always set by the g-code print file.
Pre-heat temperatures don't need to be set precisely for a specific filament. The importance is to set the most appropriate print temperature in the slicing program. 

When the nozzle is even slightly clogged, you can set a higher temperature pre-heat profile for purging/unloading, or simply set a custom temperature in "_settings > temeprature_". This process would hopefully be done quite sporadically, without an impact on everyday use.
If clogs are frequent, please review the parameters in the slicing program, or consider avoiding specifically problematic filaments. Also, feel free to contact our Customer Support for troubleshooting in case of concerns on the printer hardware side.

Our developers will look into this and make their considerations. Until then, the issue can stay open.

Michele Moramarco
Prusa Research

KaiserSoft, 2023-10-18:
I'd like to add a "Me Too" to this as the default temperature for FLEX is way too high when printing TPU. It flows out of the printer at the default temp of 240C. All TPU version I have print between 200C to 220C and usually a setting of 215C is perfect.
First layer calibration at that 240C does not work either since the filament is too runny.

I understand the concern for noobies but then please add an "Advanced User" or a "I Know what I am doing" option to the printer.

Prusa-Support, 2023-11-24:
I see. Our developers will make their considerations about it, thanks.

For now, first please tune the first layer with PLA or PETG according to your needs for the spring sheet and temperature.
Theoretically, you won't need to repeat the first layer calibration for the TPU filament after.

Also, the first layer zig-zag calibration line should be an initial reference only.
Possibly slice a 50x50x0.2 mm solid square (or anything similar) with TPU setting for fine-tuning the first layer in a standard print condition (perimeters + 45° infill).

Michele Moramarco
Prusa Research

Ferrugius, 2024-04-26:
+1 there should be some kind of "advanced settings mode" just like in pursa slicer. When you are a beginner you should not have access to some features, but there should be an option to turn them on once you gain some experience.

Prusa-Support, 2024-06-28:
They have been discussing a potential solution for this feature request in PR #3657.
However, as you can read down in that thread, there are still considerable conflicts to prevent the implementation.

Michele Moramarco
Prusa Research

Prusa-Support, 2024-11-26:
The new ["improved filament management" of FW 6.2.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases) would actually solve this issue.

Michele Moramarco
Prusa Research

closed 2024-11-26 by Prusa-Support

Razor412, 2024-12-26:
I'm new to Github so I apologize for the dumb question, but is this issue closed? This is something that should be implemented. I don't see how the suggested "improved filament management" would be a help in this situation. And just because there hasn't been any interaction on it it doesn't seem like something that should be closed just because of that. It's a pain going from PETG to PLA in a Mini.

I don't know what information is available in the printer's Why can't there be something like,

```objective-c
If oldFilamentTemperatureDefault > newFilamentTemperatureDefault Then
    MBox "Setting purge temperature at old filament default temperature."
    SetPurgeTemperature
    LoadNewFilament
Else
    LoadNewFilament
EndIf
``` 
Can we please re-open this?

CZDanol, 2025-01-02:
@Razor412 hello, this issue was about being able to set custom temperatures for filament types - which has been implemented in 6.2.0. You're talking about using a higher temperature of the two filaments during filament change. There is a different issue for that - #1821