prusa3d/Prusa-Firmware-Buddy #1259: 4.3.0 RC2 No retraction after first layer cal.

markgol, 2021-01-23:
It appears that their is not a filament retraction when the first layer cal completes.  It leaves a long string of filament from the last print point connected to the print head which continues extruding filament after reaching the park position.  Ignoring the long string of filament, it extrudes about another 3cm or more after parking.  It does not do this in the V4.2.1 release.

markgol, 2021-01-23:
I was printing with PETG.

reedy, 2021-02-04:
Seems to be the same on the 4.3.0 final release too

reedy, 2021-02-25:
4.3.1 too.

[image]

danopernis, 2024-04-17:
I can confirm that this is still an issue even in 6.0.0

reedy, 2024-06-17:
Not notably fixed...

closed 2024-08-26 by github-actions[bot]