prusa3d/Prusa-Firmware-Buddy #1262: [BUG] BSOD displayTask. During instruction prefetching, precise

peasantdave, 2021-01-26:
**Printer type** - [MINI]

**Printer firmware version**-  4.2.1-final+2072

**Original or Custom firmware** Original

**Optional upgrades** - Filament Runout Sensor and SuperPINDA

**USB drive or USB/Octoprint**
  Octoprint 1.5.3

**Describe the bug**
  Was printing fine last night. Went to start print this morning and saw "WATCHDOG RESET" "press to continue …" then would show BSOD. Shared dump below to check if this is known or new issue. 

**How to reproduce**
  Start print via Octoprint

**Expected behavior**
  Print successfully. 

**G-code**
  Happens on multiple files but before it starts printing. The "02-ARM-new6cm_0.2mm_PETG_MINI_2h45m" was able to successfully print last night. 
[06-BASE-new_0.2mm_PETG_MINI_2h41m.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5876246/06-BASE-new_0.2mm_PETG_MINI_2h41m.txt)
[02-ARM-new6cm_0.2mm_PETG_MINI_2h45m.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5876247/02-ARM-new6cm_0.2mm_PETG_MINI_2h45m.txt)

**Crash dump file**
  Attached inside the ZIP. 
[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5876250/dump.zip)

**Video**
  Pics of BSOD
[image]
[image]

zoltan-l, 2021-02-22:
HI @peasantdave 
did it happen only with the above mentioned G-code files or it appeares with any file?

peasantdave, 2021-02-22:
@zoltan-l those were the only 2 files I tested with back then. I upgraded to 4.3.0-RC2 and there was no issue printing. Currently I'm on the latest 4.3.1 and working okay still.

zoltan-l, 2021-02-23:
@peasantdave 
great to hear that our upgrade solved the issue. We keep it short time open, an if nothing apperas we will close it.

JohnnyDeer, 2021-05-13:
@peasantdave I´m closing this issue as solved.

closed 2021-05-13 by JohnnyDeer