prusa3d/Prusa-Firmware-Buddy #1264: [FEATURE REQUEST] More precise bed leveling

DhruvMadan05, 2021-01-27:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - MINI
**Optional upgrades** - None

**Is your feature request related to an existing problem? Please describe.**
When printing large or multiple objects on my MINI, I always find the outer corners of the bed the have the filament sticking less or not at all to the bed than the middle of the bed. I believe it has something to do with the small 4x4 bed leveling grid on the mini which makes the offset between points larger than what is applicable with the bed leveling. It would be good if there could be an option to change the bed leveling to a larger grid like the MK3 which has 5x5 and 7x7. 

**Describe the expected functionality**
An option in a menu that changes the mesh bed leveling grid to a larger size for more precise and accurate mesh leveling. Changes the default 4x4 to 5x5 or 6x6 similar to the MK3

Czarnyszarf, 2021-03-24:
good idea man,  has anyone picked up on this ?

loykastj, 2021-04-28:
I agree. 
Prusa, please, for the love of all that is holy, fix the bed leveling. 
I am having the same problems as DhruvMada05, the outer perimeter of the bed has horrible issues with the PETG filament sticking to the bed. 
Its driving me absolutely insane and makes the printer useless.

skarfie123, 2021-07-25:
I tried `G29 J7` to try 7x7 but it didn't do anything

HolgerMA, 2021-12-18:
/me would like this too!

Prusa-Support, 2022-12-26:
Thanks for sharing your ideas.

As emerged in #2340 (closed as duplicate), the 4x4 grid on the MINI's quite small bed is a good tradeoff: less dense than 7x7 (optional MK3 grid) but much denser than 3x3 (default MK3 grid).
A denser mesh should also consider the interference caused by magnets in such a reduced area: for example, 11 out of 49 points are affected by magnets and overridden by the firmware in the case of the MK3.

The current mesh should be sufficient and will work best on a well-straightened printer geometry.
Improving the printer's geometry after the final assembly remains our best advice.

A denser mesh on the MINI could be a bit tricky (due to the magnets) and yet it may not guarantee the best possible first layer.
It definitely sounds good though.

The issue will stay open for our developers to make their considerations on it.

Michele Moramarco
Prusa Research

closed 2024-06-17 by github-actions[bot]