prusa3d/Prusa-Firmware-Buddy #1266: [FEATURE REQUEST] Nozzle Change Feature

LuxuSam, 2021-01-28:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MINI]

**Is your feature request related to an existing problem? Please describe.**
When unscrewing the old nozzle and screwing in the new nozzle, there comes a temperature drop that throws a temperature error.

**Describe the expected functionality**
Hi, I would love to see a nozzle change feature. 

The behavior I have in mind is that first the printer unloads the filament and then the nozzle should heat up to the suggested 285°C. Afterwards there is always the problem that there comes a temperature error, as the printer cannot hold the temperature during the unscrewing of the nozzle, maybe that could be deactivated for this, as it’s really annoying if you have to restart the printer and set the temperature again.

For this feature it would also be nice if the printer prompts you with what you have to do, I.e wait for unload (usual unload behavior), wait for heat up, unscrew nozzle, wait for temperature (285°C), screw in new nozzle, wait for temp (285°) and then final tightening of new nozzle. Afterwards it could ask to load filament again.

Ondrashek, 2021-01-28:
I never had temperature error by changing nozzle. I think, that you touch to hotend thermistor by accident. Also I think that 285°C is too high from Prusa manual. I set to 250°C.

LuxuSam, 2021-01-28:
I get a thermal runway error every time I change the nozzle. When screwing in the new nozzle the temperature drops like 40-50°C and then the error occurs.

Ondrashek, 2021-01-28:
Yes, it cause hotend thermistor can't read temperature and preventive shuts temperature raising. Check your hotend thermistor cable. Chat with prusa support if you don't know. 

https://help.prusa3d.com/en/article/thermal-runaway-temperature-drops_2131

kwhansen, 2021-02-09:
A nozzle change wizard would be nice.

-find bed
-raise z to top
-unload filament
-change temp to 285
-load filament

d-zalewski, 2021-05-15:
285C seems way too high. I swapped all my nozzle at 250C just fine.

CZDanol, 2024-05-23:
Internal ticket: BFW-5524