prusa3d/Prusa-Firmware-Buddy #1267: www rozhraní na firmware Buddy_4.3.0-RC2+2729.bbf

grosiln, 2021-01-28:
Od této verze se www rozhraní po načtení buď zasekne a nebo se nenačte a zůstane na černé obrazovce
mám ve svém okolí 7x prusa mini na tomto firmware a na 5 to děla a na dvouch ne.
Předtím byl firmware prusa3d_fw_4_2_1_MINI.bbf
 Ještě nás zajímá zda je možný downgrade firmware

Ondrashek, 2021-01-28:
Downgrade FW:

Vypnout tiskárnu
Zandat USB flashku s firmwarem
Zapnout tiskárnu a po zobrazení bootloaderu zmáčknout kolečko
Vybrat Flash

zoltan-l, 2022-01-20:
@grosiln 
dělá to i v současné 4.3.4 verzi? prosím potvrdit

grosiln, 2022-01-20:
ano prostredi se snazi nacist ale to je vsepokud budete chtit poslu videotestovano na nekolika prohlizecich nekolaka pc tabletech a telefonechcekal js ze posledni aktualizace to vyresi ale nezdarilo sejeste to otestuji kdyz tiskarna natiskne to jsem na posledni verzi firmware nezkouselm.g.

closed 2024-06-17 by github-actions[bot]