prusa3d/Prusa-Firmware-Buddy #1279: [FEATURE REQUEST]Automatic unload of filament on Prusa MINI after a runout detection should happen with a delay instead of immediately, to avoid wasting filament.

anymouse404, 2021-02-01:
**If it makes sense, enter what type of printer or upgrade the feature request applies to.**
Printer type - [MINI]

**Is your feature request related to an existing problem? Please describe.**
When the filament sensor detects the filament running out, the filament is unloaded too early, wasting few centimetres of filament which would have been usable without the sensor. My print got unnecessarily interrupted on the last layer, literal seconds before it would have finished, while there was still ca. 10 cm filament left before the end would have entered the extruder.

**Describe the expected functionality**
The unload should only happen when certain length of the filament has been consumed after the runout was detected, i.e. right before the end of the filament would have entered the extruder. This will help reduce wasted filament and avoid situations where the print is unnecessarily interrupted right before the end.

jbnjohnathan, 2021-03-14:
I like this idea

Dev01-D, 2021-03-18:
I like this. i think it would be good it it paused and offered the choice of "continue" or "unload filament" as i had a run-out but had the same material ready to go. so was hoping it would pause telling me there was a run-out and then allow me to feed in my new material while using up the rest that was in the tube.

closed 2024-06-17 by github-actions[bot]