prusa3d/Prusa-Firmware-Buddy #1282: [BUG]Inconsistent extrusion width

hamlinjacob, 2021-02-02:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version**-  4.2.1

**Original or Custom firmware** Original

**Optional upgrades** - Filament Sensor, bondtech extruder

**USB drive or USB/Octoprint**
USB Drive

**Describe the bug**
 I am trying to do an extrusion multiplier test and am getting inconsistent widths on each side. I have replicated this with 6 out of 6 of my Prusa minis. I import a cube and do a vase mode to measure the extrusion width on each side. I am getting the current values (each mini is around the same) Front 0.42mm, right .44 mm, back .54mm, left .52mm. That is .12mm off.

I have tried different slicers to see if that was the issue. I have increased and decreased tension on the filament knob. I checked the tension of the belts, increased and decreased printer temps. This is consistent through different brands, type, and colors of filament.
[Mini Extrusion Mult.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/5914753/Mini.Extrusion.Mult.txt)

**How to reproduce**
  Please describe steps to reproduce the behavior.

**Expected behavior**
  With my MK3S the extrusion width is usually much closer per side. 

**G-code**
 Attached via a txt file
**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

greggree, 2021-04-01:
I can confirm that this also happens on my 2 Minis as well. The left and right sides come out at about the expected .45mm. The front comes out at .41mm and the back comes out at .49. 
[image]

greggree, 2021-04-01:
The steps to reproduce this issue can be found here: https://guides.bear-lab.com/Guide/Extrusion+multiplier+and+filament+diameter/8

hamlinjacob, 2021-04-01:
> I can confirm that this also happens on my 2 Minis as well. The left and right sides come out at about the expected .45mm. The front comes out at .41mm and the back comes out at .49.
> [image]

I am trying to figure out what is causing this issue. I am starting to think it might be firmware or axis skew. Although if it was axis skew, it is unlikely that I would have the same exact dimensions on all 6 of my minis and my dimensions match yours pretty much spot on. What upgrades do you have to the extruder or hot end? Or is it stock? Do you have any code added to the start gcode? 

It could also have something to do with the start gcode issues like #733 and #1217.

greggree, 2021-04-02:
No mods, stock 4.3.1 firmware.

JohnnyDeer, 2021-05-13:
Hi @hamlinjacob @greggree, I tested this issue on 6 stock MINI´s and result is difference between walls max 0.05mm. Could you please upload GCodes where you have this issue?

JohnnyDeer, 2021-08-02:
This issue will be closed soon for no responses.

closed 2024-06-17 by github-actions[bot]