prusa3d/Prusa-Firmware-Buddy #1285: [BUG] Prusa Connect main.js slow to load

reedy, 2021-02-04:
**Printer type** MINI

**Printer firmware version** 4.3.0

**Original or Custom firmware** Original

**Optional upgrades** Filament Runout Sensor

**USB drive or USB/Octoprint**
USB

**Describe the bug**
Prusa Connect Local Interface is slow to load, and not take over 20 seconds to load one JS file

**How to reproduce**
Load the web interface

[image]

**Expected behavior**
Interface to load in a timely fashion, especially when the printer is idle

**G-code**
n/a

**Crash dump file**
n/a

**Video**

https://user-images.githubusercontent.com/67615/106949475-e564bc80-6724-11eb-855d-153e1f93f785.mov

reedy, 2021-02-04:
And for completeness, it still takes ~20s when the printer is actively printing

stickybyte, 2021-02-10:
Can confirm that issue in version 4.3.0

turniptop, 2021-02-13:
Can confirm and add some details.  Bug appears to be browser specific.
Does not load at all with Safari.  
Loads extremely slowly on Chrome.  
Loads instantly on Firefox.

reedy, 2021-02-13:
> Does not load at all with Safari.

Ditto

> Loads extremely slowly on Chrome.

Ditto. It's slower than FF... 1.5 minutes!

[image]

Chrome 88.0.4324.150.

MS Edge ( 88.0.705.68) seems the same too...

> Loads intermittently slowly or quickly on Firefox

Seems to be the same on FF (85.0.2) for me. Sometimes it's quick,

[image]

Even doing `curl -I` can timeout... Or get a `curl: (52) Empty reply from server`

chillygithub, 2021-02-13:
I am also seeing this issue with multiple browsers on Ubuntu 20.10. In my case is is taking up to **20 minutes** to load.
On Android and Win10 I don't see the extended delay.

Reviewing captures with WireShark shows a packet sent out of order which the printer takes 46 seconds to recover from. Which is then repeated:
[image]

172.17.16.10 Prusa Mini
172.17.16.09 Ubuntu 20.10

The Mini sends packet sequence number 1073, the next sequence number should be 2145, but packet sequence number 2681 is sent. There is then a 46 second delay before the Mini sends packet 1609.

Captures have been taken both on the Ubuntu PC, and at the Mini interface with a network tap.

Captures files can be made available to Prusa staff.

Thanks for reporting this, I had been chasing the issue round my network, to ensure the issue was not my end before reporting. This issue (20 minute load time) was not apparent on 4.2.1 that was slow to load in the order of the 20 seconds reported by reedy.

reedy, 2021-02-15:
Still an issue on 4.3.1 (yes, I know it wasn't called out as being fixed, but sometimes other bug fixes creep in and issues get fixed silently).

reedy, 2021-03-18:
It's kinda interesting that people report bugs where features are rather broken in new releases, and no one has even responded/acknowledged it from Prusa. I understand you're probably busy, but you're actively committing to master for example.

Never mind as per https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.0 the feature was called out as being improved...

>Prusa Connect Local upgraded
>Prusa Connect Local has a separate language settings feature and the configuration remains saved even after a web page reload. To change the Prusa Connect interface language, connect your MINI/MINI+ to the local area network and enter its IP address into your web browser. In the top right corner, use the drop-down menu to pick the language of your choice. If you reload the page, the language setting remains saved.

Dorobuta, 2021-03-26:
> Can confirm and add some details. Bug appears to be browser specific.
> Does not load at all with Safari.
> Loads extremely slowly on Chrome.
> Loads intermittently slowly or quickly on Firefox

it would not load for me in Safari, or in Chrome - maybe I gave up in Chrome too soon.
Tried the usual stuff, ping, port scans, yada yada. Didn't put wire shark on it to watch packets.

very frustrating,. was working under safari prior to this update. I'll go back and let Chrome sit loading for a while and see if int works on my iMac or not.

toyowheelin, 2021-05-07:
This issue was happening to me today with Google Chrome. I got impatient waiting for it to load after about 5 minutes and then decided to try to do a no cache reload of the page and it came right up(ctrl+r in Chrome). I am curious if any of you have tried to do a force refresh on the page and see if it loads any quicker.

dldirector, 2021-05-20:
Force refresh makes no difference for me. I have made no sense of what is happening but here is what I can report.
I have one Linux machine in the basement, near the printer, running Firefox on Xubuntu, when I go to the Prusa Mini IP address I get the status page and all works fine.
I have another upstairs Linux machine, also Xubuntu, and trying with Chrome or Opera or Firefox, page loading shows a blank page and appears to try and load "forever".  At some point, I walked away and came back about 30 minutes later and the Prusa 
page was showing, but the print had finished and the Mini had since been turned off.  At some point, it did load.
When the page was not loading in Chrome on Linux, I opened the page on an iPad, in Chrome, and it opened immediately and seemed to work fine.
I have opened the internal console on Chrome while loading the page and I see no errors or other indicators of what is not working.
I get the same blank page, or slow loading, when a print is in progress and when the printer is completely idle.

dldirector, 2021-05-20:
See above description.  The page just loaded after 12 to 15 mins.  Attached is a screenshot of the Chrome console, network display.  Notice that main...js took 12 mins to load
[image]
 before telemetry started.

Mushmuch, 2021-06-08:
Prusa connect won't load or partially load. Safari 14.0.3 (16610.4.3.1.7)
[image]

coaxial, 2021-06-30:
This is still an issue on 4.3.1: just connected the mini to my network and none of the JS or CSS files load.

```
$ http http://192.168.1.167/
HTTP/1.0 200 OK
Content-Encoding: gzip
Content-Type: text/html
Server: Prusa Mini

<!DOCTYPE html>
<html>
  <head>
    <meta charset="utf-8" />
    <title>Original Prusa Mini - Prusa Connect</title>
  <link rel="shortcut icon" href="favicon.ico"><link href="main.b322e2869b73e803bb67.css" rel="stylesheet"></head>
  <body>
    <noscript>Your browser does not support JavaScript!</noscript>
  <script type="text/javascript" src="main.b322e2869b73e803bb67.js"></script></body>
</html>

$ http http://192.168.1.167/main.b322e2869b73e803bb67.js
HTTP/1.0 200 OK
Content-Encoding: gzip
Content-Type: application/javascript
Server: Prusa Mini

http: error: ConnectionError: HTTPConnectionPool(host='192.168.1.167', port=80): Read timed out.

$ http http://192.168.1.167/main.b322e2869b73e803bb67.js
HTTP/1.0 200 OK
Content-Encoding: gzip
Content-Type: application/javascript
Server: Prusa Mini

http: error: ConnectionError: HTTPConnectionPool(host='192.168.1.167', port=80): Read timed out.
```

reedy, 2021-08-09:
And on 4.3.2...

sgreene570, 2021-11-05:
> And on 4.3.3 too...

Same here. Can't get the web UI to properly load from any browser. @DRracer is this a known regression from any specific release?

reedy, 2021-11-05:
It wasn't an issue before upgrading to 4.3.x

Lorraine1000, 2021-11-09:
Interesting observation: Could not load the printer page in Safari or Firefox in macOS BigSur. But when I tried with Safari on my iPhone (iOS 15), it would load INSTANTLY! :-O

FileGo, 2021-11-09:
Good workaround until firmware is fixed, is to access the telemetry API directly, which gives you pretty much all the data that is available on the web interface (when it works).

It's available at `http://IP_ADDRESS/api/telemetry`.

Tobix99, 2021-12-16:
Still broken in 4.3.4

meo3000, 2022-01-04:
Solution.

Disable **Enhanced Tracking Protection** (Firefox) for the page; or allow **Insecure Content** (Edge and Chrome) in sites permissions.

Happy monitoring and happy new year!

coaxial, 2022-01-05:
> Solution.
> 
> 
> 
> Disable **Enhanced Tracking Protection** (Firefox) for the page; or allow **Insecure Content** (Edge and Chrome) in sites permissions.
> 
> 
> 
> Happy monitoring and happy new year!
> 
> 
> 
> 
> 
> 
> 
> 

This was occurring with command line clients as well and not only browsers, pointing to an issue with the server rather than the client. Are you able to curl or similar several times in a row and not have the issue anymore?

chillygithub, 2022-01-08:
> Solution.
> 
> Disable **Enhanced Tracking Protection** (Firefox) for the page; or allow **Insecure Content** (Edge and Chrome) in sites permissions.
> 
> Happy monitoring and happy new year!

Just tried, this did not resolve the issue for myself.

deltamualpha, 2022-02-06:
Chiming in to say I'm also experiencing this (and the /api/telemetry endpoint loads practically instantly, as described above, as well). Very strange behavior.

reedy, 2022-07-30:
Anyone had chance to try it on https://github.com/prusa3d/Prusa-Firmware-Buddy/tree/v4.4.0-beta2 yet?

deltamualpha, 2022-07-30:
I'll try it this weekend.

DRracer, 2022-07-30:
I'm so happy that we can finally welcome everyone to try the completely rewritten networking in 4.4-BETA2 including the new WiFi interface. On top of that, PrusaLink should be considerably more stable with many more features (not only telemetry, but G-code upload, thumbnail previews, start/stop/pause print and more).

For devs, please have a look at the PrusaLink repo (API spec): https://github.com/prusa3d/Prusa-Link-Web

deltamualpha, 2022-07-30:
Certainly seems like it's working now for me. Haven't gotten a chance to test the new features, but the raw UI loads snappily.

reedy, 2022-11-25:
Just installed https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0

It seems to work a lot better...

(Putting an API key in to view the interface feels a bit weird, but maybe that's mostly because it's being called an API key)

Prusa-Support, 2022-12-10:
Hello everyone.
I hope you are satisfied with the recent achievements and I wonder if this issue can be closed now.

Michele Moramarco
Prusa Research

closed 2023-01-10 by Prusa-Support