prusa3d/Prusa-Firmware-Buddy #1287: [BUG] "Print will end Today at..."

reedy, 2021-02-04:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** MINI

**Printer firmware version** 4.2.1, 4.3.0

**Original or Custom firmware** Original

**Optional upgrades** Filament Runout Sensor

**USB drive or USB/Octoprint**
USB

**Describe the bug**
Saying "Print will end Today at <time in past>" is not helpful after the print is finished.

**How to reproduce**
Let a print finish

**Expected behavior**
Rather than "Print will end Today at <time in past>", display something like "Print finished Today at <time in past>"; something that indicates the print is finished, and what time it was

Similar when it is cancelled; the message should change appropriately

**G-code**
N/A

**Crash dump file**
N/A

**Video**
N/A

reedy, 2021-02-04:
Please feel free to change to an enhancement as appropriate

DRracer, 2021-02-08:
good point, thank you

reedy, 2024-05-21:
Still valid.

closed 2024-07-30 by github-actions[bot]