prusa3d/Prusa-Firmware-Buddy #1289: [BUG]<Enter comprehensive title>

Af5211, 2021-02-05:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version**-  [e.g. 4.0.5, ...]

**Original or Custom firmware** [Original/Custom]

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint

**Describe the bug**
  A clear and concise description of what the bug is.

**How to reproduce**
  Please describe steps to reproduce the behavior.

**Expected behavior**
  A clear and concise description of what you expected to happen.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

Af5211, 2021-02-05:
After update Prussia Mini to 4.3 the display is going creasy he jump everywhere when I push the button. 
I have no control.

murk-sy, 2021-02-06:
You will need to provide a video because your description could mean a bug or a faulty button or something else entirely

JohnnyDeer, 2021-02-08:
Hi @Af5211, please open issue again following the template. Thank you

Af5211, 2021-02-08:
> Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.
> 
> **Printer type** - [MINI +]
> 
> **Printer firmware version**- [e.g. 4.3, ...]
> 
> **Original or Custom firmware** [Original/]
> 
> **Optional upgrades** - [Filament Runout Sensor]
> 
> **USB drive or USB/**
> Via usb
> 
> **Describe the bug**
> After update to 4.3 lcd go creasy
> 
> **How to reproduce**
> After update from 4.2.1 to 4.3 the display go everywhere by touching.
> 
> **Expected behavior**
> By 4.3 the lcd display go by touching everywhere.
> 
>  IMG_0609.MOV 
> 
> **G-code**
> Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.
> 
> **Crash dump file**
> Please attach the crash dump file. This will make it easier for us to investigate the bug.
> 
> **Video**
> Please attach a video. It usually helps to solve the problem.

https://user-images.githubusercontent.com/78602880/107198231-25fa5980-69f5-11eb-838c-cc93d90805fc.MOV

closed 2021-02-10 by JohnnyDeer