prusa3d/Prusa-Firmware-Buddy #1291: [FEATURE REQUEST] More accessable Change Steel Sheet setting.

blerrgh, 2021-02-05:
I'm very excited to see selectable steel sheet profiles in the new firmware v4.3.0

However, to change a steel sheet, it's buried behind five menus:

Settings -> HW Setup -> Steel Sheets -> <sheet profile> -> Change

I don't mind the "first layer calibration" and "reset" and eventual "rename" options buried in submenus like that (although currently "Steel Sheets" is the _only_ submenu inside the HW Setup menu which is a little silly) but from a usability standpoint I'd like to request a shortcut to just be able to change sheets right from the home screen, or maybe inside the filament change menu.

Just a simple Change Sheet -> <sheet profile>

Or Filament -> Change Sheet -> <sheet profile>

So you can change sheets in two clicks of the scroll wheel, instead of five.  Sheet selection could also be included as part of the filament change process, since people often change the build plate to match what material they're printing with.

blerrgh, 2021-02-06:
Okay - I see now that once you have multiple sheets set up, a new entry appears directly in the setup menu allowing you to quick select a sheet.

I still think that maybe that would be more appropriate in the "filament" menu or maybe even a new icon right on the main screen.  

I think the entire menu system could be optimized, honestly.  The "Info" icon is something that no one ever really needs to access, and that should be buried in the "settings" menu instead of taking up valuable icon space on the home screen.  

Even the difference between "calibration" and "settings" is kind of nebulous - for example, why is First Layer Calibration in the Calibration menu but then First Layer Calibration for different steel sheets in the Settings menu?

And in the settings menu, "factory reset" is towards the top, but really should be at the bottom.  

Overall I think the menu structures could be optimized.  I might go through them and suggest a better layout when I get a chance.

DRracer, 2021-02-08:
@blerrgh thank you for your comment and we are glad you found the switching item in settings menu, which occurs when you activate >1 sheet profiles.
The menu can be reorganized/improved of course, when adding sheet profiles we were primarily concerned about consistent user experience with the MK3/S/+ printers.

Would you mind closing this issue?

blerrgh, 2021-02-08:
As a new prusa user seeing this menu for the first time I do find it can be a little unintuitive at times.  But I understand wanting to keep the menus consistent across products.  I'll close this for now.  

Thanks!

closed 2021-02-08 by blerrgh